guino/ppsapp-rtsp #1: ppsapp patch files only

guino, 2020-12-08:
I will leave this issue open so anyone can post RTSP patch files for their ppsapp. 
The idea is to post patch files (created by users) which are small and are not property of any company which means there's no legal question on being able to share them.

The patch file format should be one supported in https://www.marcrobledo.com/RomPatcher.js/ and that is the same web site you should use to patch your ppsapp file to create a ppsapp-rtsp version of it. Please verify the MD5 of the file matches before trying to patch it or you could possibly cause damage to your device.

At a minimum we need the MD5 and the patch file but it would be nice to include details like below (which is the information from my own device):

| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c51-tuya2_bazz-2.9.6.20200422 | BE8S_H1_V10_915 | 8c03189bcfa0c691bc2f409c15a329da | Bazz Doorbell (Bell 8S) | 

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5657049/ppsapp-rtsp.zip)

guino, 2020-12-08:
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c2-weeye-1.0.0.20160101? | unknown | 671a7f2b18a289880b1bd3b4f2cafdcf | Merkury-Smart_cam-720p | 

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5657086/ppsapp-rtsp.zip)

guino, 2020-12-08:
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c51-tuya2_geeni-2.9.6 | cw017-101W-1080P | ec64bd06e807db52f50c3009827ab676 | merkury-cw017-101W-1080P | 

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5657094/ppsapp-rtsp.zip)

guino, 2020-12-08:
(Original post by @russinnes)
Firmware Version | Hardware Version | Original ppsapp MD5 | device
----------------- | ------------------|-----------------------|---------
ppstrong-c51-tuya2_xtremeecho-2.9.7.20200730|BE5S_H1_V10_915|4eb36e8475abde161c5c1290de86b88b|Smart Home Camera Bell 5S / **Energizer Connect Doorbell**

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5657113/ppsapp-rtsp.zip)

guino, 2020-12-08:
(Originl post by @adwiraguna)
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_bardi-2.9.7.20200907 | B4S_V10_H1_2063 | f4874ce6f40ff1a32e2bc1232e157105 | Bardi Outdoor Camera |

[ppsapp-rtsp-HD.zip](https://github.com/guino/ppsapp-rtsp/files/5657135/ppsapp-rtsp-HD.zip)
[ppsapp-rtsp-SD.zip](https://github.com/guino/ppsapp-rtsp/files/5657142/ppsapp-rtsp-SD.zip)

m11tch, 2020-12-08:
firmware version | hardware version | original ppsapp MD5 | device
-- | -- | -- | --
ppstrong-c51-tuya2_lcs-2.9.6.20200628 | BE8S_H1_V10_433 | 50ad9c96c65c0e446d8b3d5c8c828957 | Bell 8S (LSC Smart Connect Smart Doorbell)

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5657964/ppsapp-rtsp.zip)

guino, 2020-12-08:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_lcs-2.9.7.20201020 | BE8S_H1_V10_433 | e8900ce76220279a6e33ef65fd1d407b | Smart Home Camera (Bell 8S)  |
[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5661569/ppsapp-rtsp.zip)

guino, 2020-12-09:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_teco2-2.9.6.20200413 | BE5S_H1_V10_433 | 037ae4e296c18e585044ca54a5fdf4a9 | Bell 5S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5662976/ppsapp-rtsp.zip)

snap.cgi and mjpeg.xgi address: 0x00429c2c

lesleyvanrijn, 2020-12-13:
| Firmware Version  | Hardware Version | Original ppsapp MD5 | device |
| ------------- | ------------- |------------- |------------- |
| ppstrong-c51-tuya2_lcs-2.9.7.20201020  | BE8S_H1_V10_433  | e8900ce76220279a6e33ef65fd1d407b  | Bell 8S |

NOTE: Off-cloud patch only! No RTSP patching, since firmware 2.9.7 has ONVIF support by default. 
ONVIF service: http://IP:8000/onvif/device_service with credentials admin:admin

To enable onvif you need to block internet access (and keep it blocked), might change this later: 

Modify initrun.sh:
Add above the while loop
`if [ ! -e /mnt/mmc01/tuya_config_original.json ]; then cp /home/cfg/tuya_config.json /mnt/mmc01/tuya_config_original.json; fi`
Add inside the while loop below sleep 10:
`
if [ -e /mnt/mmc01/tuya_config.json ]; then
 cp /mnt/mmc01/tuya_config.json /home/cfg/tuya_config.json
fi
`
Boot the doorbell once, it will create the tuya_config_original.json file, copy it and rename it to tuya_config.json, open it and set onvif_enable to 1. 
[ppsapp_patched.ips.zip](https://github.com/guino/ppsapp-rtsp/files/5685177/ppsapp_patched.ips.zip)

TopdRob, 2020-12-17:
Could someone please post the correct env file for the Action LSC doorbell? I'm not sure if I have the correct one with the 37M in it.

I can't get it to work, with and without pspapp, no telnet, no rtsp, no onvif

```
bootargs=mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,

```

[image]

[image]

[image]

```
{
	"version":	1,
	"sleep_mode":	0,
	"alarm_fun_onoff":	0,
	"alarm_fun_sensitivity":	1,
	"alarm_fun_mode_switch":	0,
	"alarm_fun_time_start":	0,
	"alarm_fun_time_end":	0,
	"flip_onoff":	0,
	"light_onoff":	1,
	"night_mode":	0,
	"sound_detect_onoff":	0,
	"sound_detect_sensitivity":	0,
	"watermark_onoff":	1,
	"event_record_time":	60,
	"enable_event_record":	2,
	"record_enable":	0,
	"motion_trace":	1,
	"motion_area_switch":	0,
	"motion_area":	"",
	"motion_tracking":	0,
	"cry_detection_switch":	0,
	"humanoid_filter":	1,
	"loudspeaker_vol_pct":	100,
	"jingle_mode":	0,
	"jingle_sound":	1,
	"jingle_volume":	100,
	"jingle_exist":	0,
	"flight_bright_mode":	0,
	"flight_light_brightness":	100,
	"flight_pir_set":	0,
	"flight_pir_one":	0,
	"flight_pir_two":	0,
	"flight_pir_three":	0,
	"flight_pir_sensitivity":	0,
	"flight_alarm_fun_onoff":	0,
	"flight_on_off":	0,
	"flight_pir_light_on_time":	30,
	"flight_warn_switch":	0,
	"flight_dualbrite":	0,
	"flight_ontime":	0,
	"flight_highbrightess":	10,
	"flight_lowbrightess":	10,
	"flight_mode":	0,
	"flight_motion_sens":	0,
	"onvif_enable":	0,
	"onvif_pwd":	"admin"
}
```

http://192.168.2.183/proc/self/root/proc/cmdline
```
mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=
```

http://192.168.2.183/proc/self/root/etc/passwd
`root:$1$12345678$CTq8UQyYrE.vbbG7E8Mtj1:0:0::/root:/bin/sh`

Thanks everybody.

MokanHan, 2020-12-24:
Hi All,

I followed all instructions and cannot get Onvif or RSTP to work. Device is inaccesible from tuya app now. Can ayone be of assistance?

/home/cfg/tuya_config.json
<code>{
	"version":	1,
	"sleep_mode":	0,
	"alarm_fun_onoff":	1,
	"alarm_fun_sensitivity":	2,
	"alarm_fun_mode_switch":	0,
	"alarm_fun_time_start":	0,
	"alarm_fun_time_end":	0,
	"flip_onoff":	0,
	"light_onoff":	1,
	"night_mode":	0,
	"sound_detect_onoff":	0,
	"sound_detect_sensitivity":	0,
	"watermark_onoff":	1,
	"event_record_time":	60,
	"enable_event_record":	2,
	"record_enable":	0,
	"motion_trace":	1,
	"motion_area_switch":	0,
	"motion_area":	"",
	"motion_tracking":	0,
	"cry_detection_switch":	0,
	"humanoid_filter":	1,
	"loudspeaker_vol_pct":	10,
	"jingle_mode":	0,
	"jingle_sound":	1,
	"jingle_volume":	100,
	"jingle_exist":	0,
	"flight_bright_mode":	0,
	"flight_light_brightness":	100,
	"flight_pir_set":	0,
	"flight_pir_one":	0,
	"flight_pir_two":	0,
	"flight_pir_three":	0,
	"flight_pir_sensitivity":	0,
	"flight_alarm_fun_onoff":	0,
	"flight_on_off":	0,
	"flight_pir_light_on_time":	30,
	"flight_warn_switch":	0,
	"flight_dualbrite":	0,
	"flight_ontime":	0,
	"flight_highbrightess":	10,
	"flight_lowbrightess":	10,
	"flight_mode":	0,
	"flight_motion_sens":	0,
	"onvif_enable":	1,
	"onvif_pwd":	"admin"
}</code>

guino, 2020-12-27:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_kjell-2.9.6.20200417 | M8S_H1_V10_F23 | 27e750da764d9b9201fa12406399f758 | Smart Home Camera Mini 8S |

[ppsapp-rtsp.zip](https://github.com/guino/BazzDoorbell/files/5744903/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: N/A

guino, 2020-12-27:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_geeni-2.9.5.20200324 | M11S_H1_V10_F23 | 33cd946bc3e4004db60c612133bdd066 | Smart Home Camera Mini 11S |

[ppsapp-rtsp.zip](https://github.com/guino/BazzDoorbell/files/5744901/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x0044ed94

guino, 2021-01-03:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c5-tuya2_arlec009-2.9.0.20190808 | S5X_H1_V10_F23 | 6e807138fc0ad43f81c4b9b293b6f7cc | Orion Grid Connect Pan & Tilt Camera Model:SC009HA |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5761980/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x00481acc

guino, 2021-01-04:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a2-tuya2_teco-2.7.4.20191111 | S4C_A1_V10_1245 | 82b64e6d50aa8a2eca7d5d7eb786d284 | Nedis camera (WIFICI20CGY) |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5766651/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x002c8010

guino, 2021-01-05:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a2-tuya2_geeni-2.7.5.20200520 | M7C_AK_V10_1245 | c34616420599196a506b00b8b0e7491f | Smart Home Camera Mini 7C? |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5772116/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x002ca428

guino, 2021-01-05:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a2-tuya2_geeni-2.7.5.20200221 | M7C_AK_V10_1245 | 3be0c33cd393acfa16047e45a5e86aab | Smart Home Camera Mini 7C? |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5772560/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x002c8450

guino, 2021-01-09:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_teco2-2.9.3.20191230 | BE5S_H1_V10_433 | f4c20b9b478aeaa413ee999281ab0f4c | Bell 5S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5791500/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x0047721c

guino, 2021-01-09:
Author @exaultu
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_geeni-2.9.6.20200528 | B4S_V10_H1_2063 | a4a50713eaa055b0492d96a3022ba9b5 | Smart Home Camera (Bell 4S?) |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5791634/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x00427b3c
play.cgi address: 0x0042a5f4

guino, 2021-01-26:
Author @exaultu
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a2-tuya2_geeni-2.7.3.20190917 | M7C_AK_V10_1245 | 671a7f2b18a289880b1bd3b4f2cafdcf | Smart Home Camera (Mini 7C) |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5876447/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x002e2f30
play.cgi address: 0x002e3084

guino, 2021-01-28:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c5-tuya2_arlec009-2.9.0.20190808 | S5X_H1_V10_F23 | 6b0a73d31c4e01e700c367a570ab910c | Orion Grid Connect Pan & Tilt Camera Model:SC009HA |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5889809/ppsapp-rtsp.zip)
snap.cgi and mjpeg.cgi address: 0x00481acc
play.cgi address: 0x004824e0

guino, 2021-02-02:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_bardi-2.10.0.20201222 | B4S_V10_H1_2063 | 49b3952a658586eabd756e3b45574a69 | Bardi Outdoor Camera |

This firmware may have native onvif although reported non-functional: https://github.com/guino/BazzDoorbell/issues/20#issuecomment-771283343

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5908618/ppsapp-rtsp.zip) for RTSP on rtsp://IP:8555 (notice port 8555 instead of port 8554) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x005b97e8
play.cgi address: 0x005bd0cc

guino, 2021-02-08:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a2-tuya2_geeni-2.7.6.20210207 | M7C_AK_V10_1245 | d39b6dc4f80e3d6d8b131b5034f9e4ae | Smart Home Camera (Mini 7C) |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5944981/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x02c8d60
play.cgi address: 0x02c8eb8

guino, 2021-02-10:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_positivo-2.9.6.20200424 | S5S_H1_V10_F23 | 6db5588886b6f532f35a96db7a7156be | Smart Home Camera Speed 5S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5959497/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x0042ac2c
play.cgi address: 0x0042d6e4

guino, 2021-02-16:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_akaso-2.9.5.20200331 | B3S_V10_H1_2063 | 5222b303db8c2ff3360ea837d204a07e | Smart Home Camera (Bullet 3S) |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5989678/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x0424634
play.cgi address: 0x04270ec

guino, 2021-02-18:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a3-tuya2_merkury-4.0.6.20210207 | M11S_A2_V10_F37 | e5e559715d01cf8060d56ba97ce4a79c | Smart Home Camera (Mini 11S)  |

After patching RTSP is available at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 user **admin** and password **admin**. Onvif should be available on port 8000.

NOTE: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, there are also reports that the onvif profile has errors which prevent it from working on certain applications.

[ppsapp-onvif.zip](https://github.com/guino/ppsapp-rtsp/files/5999853/ppsapp-onvif.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x0375b1c
play.cgi request address: 0x0376328

guino, 2021-02-22:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_arlec-2.9.2.20191111 | BE5S_H1_V10_433 | 103a20358b922cf9aceeb06426e12296 | Bell 5S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6023568/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x04ee69c
play.cgi request address: 0x04ef0b8

guino, 2021-02-22:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_teco2-2.9.0.20190801 | BE5S_H1_V10_433 | 29e5ac1ce7b294ad93e848ffb67ec8b8 | Bell 5S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6023697/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x0488d4c
play.cgi request address: 0x0489760

guino, 2021-02-26:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_teco-2.9.8.20201120 | M5S_H1_V11_F23 | bcd16f22ad6d420a3d33c9ae6231a065 | Smart Home Camera (Mini 5S)  |

NOTE: This version is reported have RTSP available without patching at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 -- see: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-745618597. Onvif should be available on port 8000 (user admin and password admin).

NOTE2: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, and in that case you can try patching with the patches below:

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6052348/ppsapp-rtsp.zip) use rtsp://IP:**8555** (notice port is **8555** to not conflict with Onvif) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)

snap.cgi and mjpeg.cgi address: 0x056e7d0
play.cgi request address:  0x05720b4

guino, 2021-03-01:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_shf-2.9.1.20190907 | S4S_H1_V10_F23| aab5c31154c90af60927bd17f99d285c | Speed 4S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6063105/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x048d5e4
play.cgi request address: 0x048dff8

guino, 2021-03-12:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c4-tuya2_geeni-2.7.7.20210207 | MINI5C_V12 | 4f855b587f36c3a92842fe6683331eff | Mini 5C |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6133654/ppsapp-rtsp.zip) Required hardware programmer as described in: https://github.com/guino/Geeni720P
snap.cgi and mjpeg.cgi address: 0x03cf374 (unconfirmed, maybe: 0x03cf358)
play.cgi request address: 0x03d0504

guino, 2021-03-13:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_geeni-2.10.1.20210207 | M11S_H1_V10_F23 | a5fe2eac0f6aa506707d63558fc800b4 | Mini 11S  |

After patching RTSP is available at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 user **admin** and password **admin**. Onvif should be available on port 8000.

NOTE: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, there are also reports that the onvif profile has errors which prevent it from working on certain applications.

I also patched this one to have the old style RTSP available at rtsp://IP:8555 (notice port **8555**, no user/password)

[ppsapp-onvif.zip](https://github.com/guino/ppsapp-rtsp/files/6135730/ppsapp-onvif.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x056b6f0
play.cgi request address: 0x056efd4

guino, 2021-03-22:
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c51-tuya2_calex-2.9.6.20200409 | M7S_H1_V11_F23 | cf48c7045fbf8f0f61b239e8d38e21a7 | Mini 7S | 

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6185299/ppsapp-rtsp.zip)  please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.jpg and mjpeg.jpg address: 0x03a32c0
play.cgi request address: 0x042d6e4

guino, 2021-03-24:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a3-tuya2_merkury-4.0.2.20200929 | M11S_A2_V10_F37 | f66274e835bd4f1034dc251679bec61e | Mini 11S  |

After patching RTSP is available at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 user **admin** and password **admin**. Onvif should be available on port 8000.

NOTE: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, there are also reports that the onvif profile has errors which prevent it from working on certain applications.

[ppsapp-onvif.zip](https://github.com/guino/ppsapp-rtsp/files/6200611/ppsapp-onvif.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x037b3ac
play.cgi request address: 0x037bb1c

JonSingleton, 2021-03-26:
> Author @exaultu
> 
> Firmware Version	Hardware Version	Original ppsapp MD5	device
> ppstrong-c51-tuya2_geeni-2.9.6.20200528	B4S_V10_H1_2063	a4a50713eaa055b0492d96a3022ba9b5	Bullet 4S
> [ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5791634/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
> snap.cgi and mjpeg.cgi address: 0x00427b3c
> play.cgi address: 0x0042a5f4

Hey Guino,

I have the Mercury 1080p outdoor camera from Walmart, matching this bad boy entirely. The mjpeg / snap as well as the play addresses all matched and verified they're working, but I can't seem to get the ppsapp provided to work for me. I did verify the md5 matched the one your hack grabbed me in home/app/ppsapp, I downloaded the patch as a zip, and uploaded it (still zipped) along with my ppsapp

My original ppsapp: [ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6210670/ppsapp.zip)
The patch file I used: 
[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6210681/ppsapp-rtsp.zip)
The ppsapp placed at the sd card's root:
[ppsapp-patched.zip](https://github.com/guino/ppsapp-rtsp/files/6210697/ppsapp-patched.zip)

http://admin:056565099@ip/proc/cmdline
	mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd

http://admin:056565099@ip/devices/deviceinfo
	{
	  "devname": "Smart Home Camera",
	  "model": "Bullet 4S",
	  "serialno": "*",
	  "softwareversion": "2.9.6",
	  "hardwareversion": "B4S_V10_H1_2063",
	  "firmwareversion": "ppstrong-c51-tuya2_geeni-2.9.6.20200528",
	  "authkey": "*",
	  "deviceid": "*",
	  "identity": "*",
	  "pid": "aaa",
	  "WiFi MAC": "*"
	}

I jumped in via telnet and ran:
`(PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  kill $PPSID
  /mnt/mmc01/ppsapp &> /mnt/mmc01/ppsapp.log &)`

And watched for anything RTSP related. No mention of starting any RTSP server - I also noticed a couple of errors. I've attached what seems relevant but I really can't tell if the constant "DEBUG tuya_network_manager.c:329] driver_set_freq wifi_config_is_stoped" followed by switching to a new channel so frequently (few times a second in some cases).

ppsapp log: [ppsapp.log](https://github.com/guino/ppsapp-rtsp/files/6210640/ppsapp.log)

I have tried three separate SD cards (I've been known to get some cheapo cards), no bueno. Any I-clue what might be giving me heartburn on this one?

guino, 2021-03-29:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_shf-2.9.1.20190907 | ? | d45f47d6ef4e4ce3b8cac55085e7773d | ? |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6222869/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x048a4fc
play.cgi request address: 0x048af10

Lollo9191, 2021-03-31:
Hi, can you tell me where found patch file for:
softwareversion":"2.9.0","hardwareversion":"BE5S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_life-2.9.0.20190727"
Thanks!!

guino, 2021-04-05:
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c51-tuya2_calex-2.9.6.20200522 | M7S_H1_V11_F23 | c035aef31d4ba53015af696769f336cf | Mini 7S | 

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6260349/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.jpg and mjpeg.jpg address: 0x042ac34
play.cgi request address: 0x042d6ec

guino, 2021-04-07:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_panam-2.9.2.20191107 | B2S_V10_H1_2063 | 7ffb374de97a9802aba1150a7fbc0f33 | Bullet 2S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6268637/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x04e53bc
play.cgi address: 0x04e5dd8

guino, 2021-04-08:
(Original post by @russinnes -- **Energizer Connect Doorbell**)
Firmware Version | Hardware Version | Original ppsapp MD5 | device
----------------- | ------------------|-----------------------|---------
ppstrong-c51-tuya2_xtreme-2.9.7.20200730 | BE5S_H1_V10_915 | ba34e6f9e375e0c962b02133434ce55a | Bell 5S

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6280358/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.jpg and mjpeg.jpg address: 0x046f7f4
play.cgi request address: 0x04722ac

guino, 2021-04-08:
(If you have this one you likely can update your device -- please check before doing the work)
Firmware Version | Hardware Version | Original ppsapp MD5 | device
----------------- | ------------------|-----------------------|---------
ppstrong-c51-tuya2_xtreme-2.8.0.20190725 | BE5S_H1_V10_915 | 6e97e7a6dae966eeaf3c4629c678f3b7 | Bell 5S

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6280433/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.jpg and mjpeg.jpg address: 0x0480a54
play.cgi request address: 0x0481468

guino, 2021-04-28:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_teco-2.9.2.20191106 | B2S_V10_H1_2063 | 2e9eb43c6cffce86275b175498d182b1 | Bullet 2S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6393139/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x04ee69c
play.cgi address: 0x04ef0b8

guino, 2021-04-29:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_istar-2.9.7.20200814 | BE5S_H1_V10_433 |  54c58b61771c65ed1f870385cd77c7d7 | Bell 5S  |

NOTE: This version should have RTSP available without patching at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 -- see: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-745618597. Onvif should be available on port 8000 (user admin and password admin).

NOTE2: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, and in that case you can try patching with the patch below:

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6401972/ppsapp-rtsp.zip) use rtsp://IP:**8555** (notice port is **8555** to not conflict with Onvif) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x05504b4
play.cgi request address:  0x0552f6c

guino, 2021-05-01:
(Originl post by @adwiraguna)
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a3-tuya2_merkury-4.0.0.20200903 | B4S_V10_A2_2063 | d67a5a242df168fd9997e5d364f0e53c | Bullet 4S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6408138/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x027f5f8
play.cgi request address: 0x027fa58

guino, 2021-05-03:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-a2-tuya2_geeni-2.7.6.20210207 | M7C_AK_V10_1245 | d39b6dc4f80e3d6d8b131b5034f9e4ae | Mini 7C |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6417933/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x02c8d60
play.cgi request address: 0x02c8eb8

danielsan79, 2021-05-05:
I bought this camera from some onlinewebshop, could you take a look? onvif  is available. Is there a patch?

model | "Speed 3S"
serialno | "063159911"
softwareversion | "2.9.7"
hardwareversion | "S3S_H1_V10_F23"
firmwareversion | "ppstrong-c51-tuya2_laxihub-2.9.7.20201028"
authkey | "4O2ATnzrO7CGMFD2ALp2SYxbC0yY1EXK"
deviceid | "pp01e81b775a8b26c82f"
identity | "MR2008221200901349"
pid | "aaa"

guino, 2021-05-06:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_laxihub-2.9.7.20201028 | S3S_H1_V10_F23 | a4cd739334efbb719f21454802dce9d2 | Speed 3S |

NOTE: This version should have RTSP available without patching at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 -- see: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-745618597

NOTE2: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, and in that case you can try patching with the patch below:

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6435777/ppsapp-rtsp.zip) use port **8555** (rtsp://IP:8555) as ONVIF may already be using port 8554, please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x0551634
play.cgi request address: 0x05540ec

stack0423, 2021-05-12:
Hi,can any one help me with the following.
I did my doorbell with the hack that works, when, I put the test url in it says done.
now want me to make mjpeg work, for this doorbell
ppstrong-c51-tuya2_lcs-2.9.7.20201020
BE8S_H1_V10_433 
e8900ce76220279a6e33ef65fd1d407b Bell 8S

in the mjpeg.cdi place i this address: 00551634
when i open the mjpeg stream i get the 404 error.
I don't know if I am entering the correct address in the correct line.
There is no need to patch with version 2.9.7, is it?

i also don't need the rtsp stream just mjpeg.

Thanks

guino, 2021-05-15:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_akaso-2.9.2.20191108 | B2S_V10_H1_2063 | 609b109497dba30edc69fded30070dca | Bullet 2S |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6488233/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x04e84ac
play.cgi address: 0x04e8ec8

guino, 2021-05-17:
(Orion Grid Connect Pan & Tilt Camera Model SC009HA)
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c5-tuya2_arlec009-2.9.6.20200414 | S5X_H1_V10_F23 | ebdf087b5cd384eadea3e921f0948f2e | Speed 5X |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6496119/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080) -- REQUIRES 
snap.cgi and mjpeg.cgi address: 0x00434a0c
play.cgi address: 0x004374c4

guino, 2021-05-18:
(Orion Grid Connect Pan & Tilt Camera Model SC009HA)
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c5-tuya2_arlec009-2.9.6.20200702 | S5X_H1_V10_F23 | e2cc37a3b73bac14c0df2d5ad9edb110 | Speed 5X |

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6502238/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x0047e71c
play.cgi address: 0x004811d4

liberty4fr, 2021-08-10:
Hi,

Can you please patch my psapp ? Thanks
Smart Home Camera","model":"Mini7S"
"serialno":"058193465"
"softwareversion":"2.9.1"
"hardwareversion":"M7S_H1_V11_F23"
"firmwareversion":"ppstrong-c51-tuya2_shf2.9.1.20190907"
"authkey":"vn1tnO6JgBKh7ZuKICp7mHFSytLQcnVK"
"deviceid":"shft4c86cbdb8b89c85e"
"pid":"aaa","WiFi MAC":"c4:3a:35:21:5a:d7"}

Tyrian81, 2021-08-23:
I have hacked version 2.9.7 got an mqtt logparser on it as well , Tuya app is giving me a quite annoying update notification, can i safely update it or do i need to hack all over again .. ?
[image]

obisidianouc, 2022-01-19:
I would like to ask whether there is a patch for ppstrong-a3-tuya2_merkury-4.0.7.20210624? The structure is very different and I am lost.

The device info is below

{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"102785221","softwareversion":"4.0.7","hardwareversion":"M11S_A2_V20_GC1","firmwareversion":"ppstrong-a3-tuya2_merkury-4.0.7.20210624","identity":"M720016K7N02000396","authkey":"OTYFdMZ6TQUjxsYdqEGR7iBSt0AMgxlD","deviceid":"pp01c742cbe8597ce82e","pid":"aaa","WiFi MAC":"84:7a:b6:f4:6d:a2","ETH MAC":"00:00:00:00:00:00"}

abates-asasolutions, 2022-05-31:
Firmware Version | Hardware Version | Original ppsapp MD5 | device
-- | -- | -- | --
ppstrong-a2-tuya2_geeni-2.7.8.20210514 | M7C_AK_V10_1245 | b6243b45108ccc5299f560301ae07d6c | Mini 7C

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/8801800/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)

snap.cgi and mjpeg.cgi address: [Unknown, similar issue to #38 ]
play.cgi address: [Unknown, similar issue to #38 ]

The 2.7.8 firmware is notable for having been compiled with symbols, so in many (most?) instances we can see the original Tuya function names instead of guessing them.

It required an instruction change to always start RTSP in the Echo Show function, but did already call the Echo Show function and so didn't require a instruction change to override a function call.

andrewjbates, 2022-05-31:
Firmware Version | Hardware Version | Original ppsapp MD5 | device
--- | --- | --- | ---
ppstrong-a2-tuya2_geeni-2.7.8.20210514 | M7C_AK_V10_1245 | b6243b45108ccc5299f560301ae07d6c | Mini 7C

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/8801800/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)

snap.cgi and mjpeg.cgi address: [Unknown, similar issue to #38 ]
play.cgi address: [Unknown, similar issue to #38 ]

The 2.7.8 firmware is notable for having been compiled with symbols, so in many (most?) instances we can see the original Tuya function names instead of guessing them.

It required an instruction change to always start RTSP in the Echo Show function, but did already call the Echo Show function and so didn't require a instruction change to override a function call.

andrewjbates, 2022-06-02:
@guino [ppstrong-a2-tuya2_geeni-2.7.8.20210514.zip](https://github.com/guino/ppsapp-rtsp/files/8826780/ppstrong-a2-tuya2_geeni-2.7.8.20210514.zip)

SoloAsh, 2022-07-06:
Just wanted to see if a fix for this issue has been resolved.-
Above already exists a patch for "ppstrong-c51-tuya2_geeni-2.10.1.20210207"
Device info-{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"060343649","softwareversion":"2.10.1","hardwareversion":"M11S_H1_V10_F23","firmwareversion":"ppstrong-c51-tuya2_geeni-2.10.1.20210207","authkey":"lozPNpSAnbJwVhz7GKp4iNV5l5UpW9I9","deviceid":"pp0132b1202c630af158","identity":"MR2005150400913973","pid":"aaa","WiFi MAC":"38:01:46:1e:4d:b3"}

ZIP file is provided above but like the post says Some are reporting rtsp and onvif still not available and this is my issue. (RTSP is needed)
I repatched my ppa with the Rom site and provided zip.
The hack reported "done" with no problem though so I was happy on this :)
I would appreciate any fix or info you could share.

Thank you for your hard work on this whole project it is very much appreciated and I have learned a lot.

dmidthun, 2022-09-26:
Bought some new cams today, out of the box the root "half" works: 

{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"105202679","softwareversion":"5.2.5","hardwareversion":"M11S_A5_V10_MIS1","firmwareversion":"ppstrong-a5-tuya2_general-5.2.5.20211201","identity":"","authkey":"DAAtLEvS6SxPQxN6VuvXTufTeLe0s66A","deviceid":"tstjb6f11613b631505d","pid":"aaa","WiFi MAC":"9c:1c:37:4c:a7:ae","ETH MAC":"9c:1c:37:4c:a7:ae"}

looking to get rtsp / onvif for these guys

Thanks Guino!!

RomchikL, 2022-09-30:
Hello!

Could you clarify the address for snap/mjpeg for the camera:
{"devname":"Smart Home Camera","model":"Mini 7S","serialno":"100372114","softwareversion":"4.0.6","hardwareversion":"M7S_A2_V10_F37","firmwareversion":"ppstrong-a3-tuya2_laxi-4.0.6.20210326","identity":"M1F0079H3H01700027","authkey":"XXX","deviceid":"pp0184b6ab1f26a1312f","pid":"aaa","WiFi MAC":"XXX","ETH MAC":"XXX"}

There are similar fw in the table, but I'm not sure if they are correct.

sle118, 2023-04-21:
<html>
<body>

firmware version | hardware version | original ppsapp MD5 | device
-- | -- | -- | --
ppstrong-c51-tuya2_bazz-2.10.0.20210801 | BE8S_H1_V10_915 | ad654f758e11da31a077130709c2a456| Bell 8S

<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(31, 35, 40); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">         <a href="https://github.com/guino/ppsapp-rtsp/files/11295098/ppsapp-rtsp.zip"            style="box-sizing: border-box; background-color: transparent; color: var(--color-accent-fg); text-decoration: none;">ppsapp-rtsp.zip</a><span> </span>please        use steps above to patch it.<br style="box-sizing: border-box;">snap.cgi and mjpeg.cgi address: 0x0056e6b0       <br            style="box-sizing: border-box;">play.cgi address: 0x571f94  </p>
</body>
</html>

[3 more replies]