guino/ppsapp-rtsp #2: > Author @exaultu

guino, 2021-03-26:
> Author @exaultu
> 
> Firmware Version	Hardware Version	Original ppsapp MD5	device
> ppstrong-c51-tuya2_geeni-2.9.6.20200528	B4S_V10_H1_2063	a4a50713eaa055b0492d96a3022ba9b5	Bullet 4S
> [ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/5791634/ppsapp-rtsp.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
> snap.cgi and mjpeg.cgi address: 0x00427b3c
> play.cgi address: 0x0042a5f4

Hey Guino,

I have the Mercury 1080p outdoor camera from Walmart, matching this bad boy entirely. The mjpeg / snap as well as the play addresses all matched and verified they're working, but I can't seem to get the ppsapp provided to work for me. I did verify the md5 matched the one your hack grabbed me in home/app/ppsapp, I downloaded the patch as a zip, and uploaded it (still zipped) along with my ppsapp

My original ppsapp: [ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6210670/ppsapp.zip)
The patch file I used: 
[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/6210681/ppsapp-rtsp.zip)
The ppsapp placed at the sd card's root:
[ppsapp-patched.zip](https://github.com/guino/ppsapp-rtsp/files/6210697/ppsapp-patched.zip)

http://admin:056565099@ip/proc/cmdline
	mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd

http://admin:056565099@ip/devices/deviceinfo
	{
	  "devname": "Smart Home Camera",
	  "model": "Bullet 4S",
	  "serialno": "*",
	  "softwareversion": "2.9.6",
	  "hardwareversion": "B4S_V10_H1_2063",
	  "firmwareversion": "ppstrong-c51-tuya2_geeni-2.9.6.20200528",
	  "authkey": "*",
	  "deviceid": "*",
	  "identity": "*",
	  "pid": "aaa",
	  "WiFi MAC": "*"
	}

I jumped in via telnet and ran:
`(PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  kill $PPSID
  /mnt/mmc01/ppsapp &> /mnt/mmc01/ppsapp.log &)`

And watched for anything RTSP related. No mention of starting any RTSP server - I also noticed a couple of errors. I've attached what seems relevant but I really can't tell if the constant "DEBUG tuya_network_manager.c:329] driver_set_freq wifi_config_is_stoped" followed by switching to a new channel so frequently (few times a second in some cases).

ppsapp log: [ppsapp.log](https://github.com/guino/ppsapp-rtsp/files/6210640/ppsapp.log)

I have tried three separate SD cards (I've been known to get some cheapo cards), no bueno. Any I-clue what might be giving me heartburn on this one?

_Originally posted by @JonSingleton in https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-808069453_

guino, 2021-03-26:
@JonSingleton I will take a look at your files and let you know. I created an issue for this as you posted in a section reserved for patch files.

JonSingleton, 2021-03-26:
I could have sworn I clicked "Reference in new issue" - I suck. Thanks for offering to look at it. If there's any other info you need don't hesitate I have it plugged in still and it's just a telnet connection away - or shoot me a message and I can open up and forward the telnet port and message you the IP:23 of the camera if you'd prefer.

guino, 2021-03-26:
@JonSingleton Everything seems to be in order with your files/patch but your device doesn't seem to be getting online. Are you running this on a local network (or vlan) without internet access ? this is not an 'offline' patch so the device still needs internet access to connect to Tuya servers to be able to synchronize time so it can finish booting up completely (to the point where it will enable RTSP). 

If you want/need to run this offline you can try this patch: [ppsapp-offline.zip](https://github.com/guino/ppsapp-rtsp/files/6212832/ppsapp-offline.zip) but doing that will require you to do something like this: https://github.com/guino/BazzDoorbell/issues/4#issuecomment-744084557 to set the time when the device boots up.

JonSingleton, 2021-03-26:
I'll be damned. I was so tired when I gave up to write that issue my brain just wasn't firing on all cylinders. 

You mentioning it not being able to access the network pointed me in the right direction. I'm sure it's been written somewhere and I just didn't happen to read it while skimming through everything - I was in such a hurry to get it rooted and start messing with it that I just didn't register it to my geeni app. It only makes sense that without having been synced to the app, the network settings were never transferred to the camera to make that initial connection. smh.

Thanks for digging through that. Can confirm RTSP is running now and the patch file can be marked as working still.
[image]
Please excuse my messy woodshop and painting station. A clean shop is an unused shop I always did say.

guino, 2021-03-26:
@JonSingleton Thanks for the feedback! Glad you got it working - I saw the device was connecting to a network (log file) so I thought it was registered in the app but I suppose you may have used the ppsFactoryTool.txt file instead.