guino/ppsapp-rtsp #4: Request: ppstrong-c51-tuya2_teco-2.9.2.20191106

Suspect-byte, 2021-04-28:
Hi @guino,

I am back with another camera. Can you maybe patch the ppsapp?

`{"devname":"Smart` Home Camera",
"model":"Bullet 2S",
"serialno":"058960514",
"softwareversion":"2.9.2",
"hardwareversion":"B2S_V10_H1_2063",
"firmwareversion":"ppstrong-c51-tuya2_teco-2.9.2.20191106",
"authkey":"L8eghnyzhgbxkDjWjY7o1jVtvInnmARi",
"deviceid":"pp0136a7872d0cf0d95c",
"identity":"MR1909090400900528",
`"pid":"aaa",`

[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6391748/ppsapp.zip)

Thanks in advance!

guino, 2021-04-28:
I should be able to do it today.

guino, 2021-04-28:
Here's the patch:
https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-828564836

Let me know if you have any issues (I don't expect any).

Suspect-byte, 2021-04-28:
@guino Thank you so much for the fast response! 
Cheers!

closed 2021-04-28 by Suspect-byte