guino/ppsapp-rtsp #5: ppstrong-c51-tuya2_istar-2.9.7.20200814

gijsje, 2021-04-29:
Hi Guino,

I have this doorbell and there is no patch availible yet.
Could you have a look at this?

```
{"devname":"Smart Home Camera",
"model":"Bell 5S",
"serialno":"061638300",
"softwareversion":"2.9.7",
"hardwareversion":"BE5S_H1_V10_433",
"firmwareversion":"ppstrong-c51-tuya2_istar-2.9.7.20200814",
"authkey":"D6lhAACmnhD3E9CgSPxgFsEpHOC3k7fT",
"deviceid":"pp011202d89fc2746456",
"identity":"MR2007180100100494",
"pid":"aaa",
"WiFi MAC":"changed"}
```

MD5 is 
54c58b61771c65ed1f870385cd77c7d7

I do not know if you anything else if you do please let me know.
Thanks in advance

[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6401724/ppsapp.zip)

guino, 2021-04-29:
I'm looking at it right now

guino, 2021-04-29:
@gijsje Did you try using onvif by setting onvif_enable on the tuya_config.json file ? it looks like this version should work with it without patching.

guino, 2021-04-29:
Here's the patch to enable RTSP on port 8555 (not 8554) -- it should work with or without enabling onvif (which has its own RTSP on port 8554):

https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-829636432

gijsje, 2021-04-30:
Guino Thanks a lot i have it working now
Coffee/beer donation on its way

guino, 2021-04-30:
@gijsje much appreciated! Cheers!