guino/ppsapp-rtsp #7: I bought this camera from some onlinewebshop, could you take a look? onvif  is available. Is there a patch?

guino, 2021-05-05:
I bought this camera from some onlinewebshop, could you take a look? onvif  is available. Is there a patch?

model | "Speed 3S"
serialno | "063159911"
softwareversion | "2.9.7"
hardwareversion | "S3S_H1_V10_F23"
firmwareversion | "ppstrong-c51-tuya2_laxihub-2.9.7.20201028"
authkey | "4O2ATnzrO7CGMFD2ALp2SYxbC0yY1EXK"
deviceid | "pp01e81b775a8b26c82f"
identity | "MR2008221200901349"
pid | "aaa"

_Originally posted by @danielsan79 in https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-832991785_

guino, 2021-05-05:
@danielsan79 There's no patch for this ppsapp yet - just post a zip of the ppsapp here in github and I can patch it for you.

danielsan79, 2021-05-06:
https://www90.zippyshare.com/v/VZKrKzT1/file.html

I hope it will work on my synology nas!

danielsan79, 2021-05-06:
No luck, no fun time :( 

Onvif is stil not working, synology supports (also) RTSP but also have no connection.
I also use the patch it will connect the snap.cgi and mjpeg.cgi 
but not play.cgi request address: 0x05540ec.

of course i use diffrent players and clients.

guino, 2021-05-06:
@danielsan79 does rtsp work anywhere else (vlc, phone app, etc)? Please note the rtsp on port 8555 from the patched version uses no user/password. 

How did you try play.cgi ? that is for playing audio files so it requires parameters and won’t work from an application like vlc, etc. very few people use this feature.

danielsan79, 2021-05-07:
Yes, little fun time!

I can acces the RTSP addres but, after exactly 19 minutes the camera reboot itself, with the annoying boot sounds. ONVIF doesnt work. I do block all connection with pihole, but it doesnt seems to matter if i dont.
 any advise?

guino, 2021-05-07:
@danielsan79 can you try without blocking connections? The patch I made is not meant to work “offline”.

The 19 minutes reboot issue you are describing is a common symptom of using UDP instead of TCP for the RTSP stream. Just adjust the settings in VLC (or app) and it should stop rebooting.

danielsan79, 2021-05-07:
It was the UDP, so i config the client and it work! ONVIF may not be supported.

RTSP is working on my nas, i blocked all traffic from the camera to WAN, only control (the camera) inside the network, and use vpn server to connect my network, when im outside. Thanx for the help, coffee is on the way!

guino, 2021-05-07:
@danielsan79 glad you got it working! Thanks for the feedback and for the coffee! Cheers!