guino/ppsapp-rtsp #8: Hi,can any one help me with the following.

guino, 2021-05-12:
Hi,can any one help me with the following.
I did my doorbell with the hack that works, when, I put the test url in it says done.
now want me to make mjpeg work, for this doorbell
ppstrong-c51-tuya2_lcs-2.9.7.20201020
BE8S_H1_V10_433 
e8900ce76220279a6e33ef65fd1d407b Bell 8S

in the mjpeg.cdi place i this address: 00551634
when i open the mjpeg stream i get the 404 error.
I don't know if I am entering the correct address in the correct line.
There is no need to patch with version 2.9.7, is it?

i also don't need the rtsp stream just mjpeg.

Thanks

_Originally posted by @stack0423 in https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-839947501_

guino, 2021-05-12:
@stack0423 please post the URL you're using along with username and password you're using.
In order to use mjpeg you need to have the mmc files correctly placed in the SD card (not just mjpeg.cgi) -- namely you need: busybox (which whould be over 1Mb in size), httpd.conf (with your user/password defined in it) and the cgi folder/directory with the files in it.

Feel free to upload a zip of your SD card files (except SDT folder/directory) to this github issue and I can take a look.

stack0423, 2021-05-12:
Thanks for your quick response, I use the following url I disabled username and password in httpd with #

http://192.168.178.51:8080/cgi-bin/mjpeg.cgi
also my root files
[upload.zip](https://github.com/guino/ppsapp-rtsp/files/6468067/upload.zip)

guino, 2021-05-12:
@stack0423 your seem fine -- the URL should have the user/password so you should try: http://admin:admin@192.168.178.51:8080/cgi-bin/mjpeg.cgi

I would also try (just to see if the web server is working at all):
http://admin:admin@192.168.178.51:8080/upload.html

I understand you have not patched ppsapp so just make sure that you **boot** the device with the SD card and files already in it.

stack0423, 2021-05-13:
Still 404, upload works. I see my folders that I don't see a home folder, is that correct?
I try to reset doorbell without sdcard but then won't boot.
note! I have previously updated this doorbell from 2.9.6. to 2.9.7

stack0423, 2021-05-14:
I know what went wrong,modified mjpeg.cgi with wrong editor now use notepad ++  and now works.