guino/ppsapp-rtsp #10: Please patch my ppsapp file

DanieduPlessis, 2021-05-15:
Hi

Can you please patch my ppsapp file.
Here is the info from the deviceinfo
{
"devname": "Smart Home Camera",
"model": "Bullet 2S",
"serialno": "057634187",
"softwareversion": "2.9.2",
"hardwareversion": "B2S_V10_H1_2063",
"firmwareversion": "ppstrong-c51-tuya2_akaso-2.9.2.20191108",
"authkey": "PxnBwpJPShMeELyR3rVlAyqojZkHFtRR",
"deviceid": "pp0139e49fa68b2e5fb3",
"pid": "aaa",
"WiFi MAC": "ac:5d:5c:0b:9f:ad"
}

and attached the original ppsapp file

Thanks
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6487984/ppsapp.zip)

guino, 2021-05-15:
@DanieduPlessis I made the patch and posted it here: https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-841726700 

Let me know if you have any issues but I expect it to work just fine.

DanieduPlessis, 2021-05-16:
Thanks much appreciated, the mjpeg stream is working but the rtsp feed is not working, attached is a zip od the sd card

Thanks
Danie
[sd card backup.zip](https://github.com/guino/ppsapp-rtsp/files/6489357/sd.card.backup.zip)

guino, 2021-05-16:
@DanieduPlessis your files seem to be correct, so as long as you're booting the camera with the SD card already in place it should work (you should hear 2 startup sounds and then try RTSP after the 2nd startup sound). Can you tell me how you're trying the RTSP ? It should just be rtsp://IP:8554 without a user/password -- you have to use a phone app or some application that supports RTSP (can't be on the web browser). There are some reports that VLC for MAC doesn't always work, so I recommend trying a phone app or some other machine (since you appear to be using a MAC(.

DanieduPlessis, 2021-05-17:
@guino yes I'm booting the camera with the SD card and there is 2 startup sounds, I'm also trying to connect to the rtsp stream using apps on multiple devices, all iOS devices. I'm not using a username or password and the connection url is rtsp://192.168.4.5:8554
ITs not a mayor issue, as I'm currently using the mjpeg stream, and it is working 100% for what I want to do

Thanks
Danie

guino, 2021-05-17:
@DanieduPlessis I did double check the changes I made and the changes on your file so I am not sure why it isn't working. If you ever want to figure it out I would need you to  modify this line in custom.sh:
```
/mnt/mmc01/ppsapp &
```
to be:
```
/mnt/mmc01/ppsapp 2>&1 > /mnt/mmc01/ppsapp.log &
```
Then boot your camera with SD card in it, try to access RTSP from anywhere and then send me the ppsapp.log file that will be created in the SD card.

Obviously if you're happy as-is you don't have to do anything else, in fact, if you just want to use mjpeg you don't even need the patched ppsapp in the root of the SD card (The built-in one should work with it).