guino/ppsapp-rtsp #11: New ppsapp file to please patch - firmware ppstrong-c5-tuya2_arlec009-2.9.6.20200414

BeHappy2021, 2021-05-17:
Hi Guino,
Please can you patch the attached firmware for my Orion Smart HD Pan and Tilt camera
Device info is:

{
  "devname": "Smart Home Camera",
  "model": "Speed 5X",
  "softwareversion": "2.9.6",
  "hardwareversion": "S5X_H1_V10_F23",
  "firmwareversion": "ppstrong-c5-tuya2_arlec009-2.9.6.20200414",

[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6496017/ppsapp.zip)
}

MD5 is ebdf087b5cd384eadea3e921f0948f2e

Looking to get access to the jpeg snapshot and rtsp stream if possible. Thanks in advance.

guino, 2021-05-17:
Here's the RTSP patch and addresses: https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-842546704
I don't expect you to have any issues.

BeHappy2021, 2021-05-18:
Hi,
Thanks for being so quick. Something is not quite right. I've changed the addresses and done patch. Seems like the camera is in a bootloop as every few minutes I hear the start up sound. When I try http://user:pass@ip:8080/cgi-bin/snap-cgi I get a page with the following:
The image "http://ip:8080/cgi-bin/snapcgi" cannot be displayed, because it contains errors.
Tried accessing with the Tuya app and it gets passed the bit of 'established encryted connection' but gets stuck on 'retrieving video stream'. I once tried just after hearing the start-up sound and got it got the stream for a couple of seconds before losing it and the start-up sound played again.
Anything to try next?

guino, 2021-05-18:
@BeHappy2021 this sounds like an issue we have seen before and I thought was related to the firmware version but since you seem to be having this issue on 2.9.6 I can only assume the issue is linked with the hardware type (speed 5x).

if you just want mjpeg/snap, you can simply remove the patched ppsapp from the root of the SD card and the camera should work normally and allow snap/mjpeg to work (no rtsp).

if you want to to usr RTSP it seems like you would have to do some additional steps to make it work — these steps are listed here: https://github.com/DanTLehman/orion_sc008ha#how-did-i-get-it-working

Let me know if you make any progress.

BeHappy2021, 2021-05-18:
Hi Guino,

Thanks for the extra help and sorry for wasting your time but I think in my case is was down to user error. Originally I started off from this page: [https://github.com/guino/BazzDoorbell/issues/2](url) and then as I didn't have a '#' on the MTDNUM line chose to go to [https://github.com/guino/BazzDoorbell/issues/13](url) as it looked easier. It ended up in what seemed like the bootloop.

I tried that a couple of times and then put everything back to normal and started from the beginning just following [https://github.com/guino/BazzDoorbell/issues/2](url) all the way through and it worked! I think partly the errors were down to me mixing files from ppshack.zip and ppshack2.zip with switching between instructions so having an env which wasn't quite correct.

Thank you so much. Now to go buy you a well deserved coffee!

guino, 2021-05-18:
@BeHappy2021 glad you got it working, thanks for the feedback!

BeHappy2021, 2021-05-18:
Hello again,
Now trying to do this on a second camera. Was hoping all would be the same but it isn't. Details for this camera are:

{
"devname": "Smart Home Camera",
"model": "Speed 5X",
"softwareversion": "2.9.6",
"hardwareversion": "S5X_H1_V10_F23",
"firmwareversion": "ppstrong-c5-tuya2_arlec009-2.9.6.20200702",
}

MD5 = e2cc37a3b73bac14c0df2d5ad9edb110

Can you do another patch please?

guino, 2021-05-18:
@BeHappy2021 I can patch it, just need you to post the ppsapp file here in github.

BeHappy2021, 2021-05-18:
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6502173/ppsapp.zip)
Whoops sorry, I zipped it and forgot to attach it!

guino, 2021-05-18:
@BeHappy2021 here you go: https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-843243826

BeHappy2021, 2021-05-18:
Thanks. Worked first time so I must have just got it wrong with the other camera. Both working perfectly now!

guino, 2021-05-18:
@BeHappy2021 awesome! Enjoy and thanks for the coffee!