guino/ppsapp-rtsp #12: bootloop with patched ppsapp

datNoko, 2021-05-20:
Hi,
My Camera is the:
ppstrong-c51-tuya2_lcs-2.9.6.20200628 | BE8S_H1_V10_433 | 50ad9c96c65c0e446d8b3d5c8c828957 | Bell 8S
MD5 matches with my stock ppsapp

i have the following problem:
when i put the patched ppsapp file on my mmc root folder my camera doesn't boot and appears to be stuck in a bootloop.
I hear the startup sound probably every minute, the led is red all the time and the device isn't available in my network (aka no ping).
When i remove the ppsapp from my sdcard the cam is booting up normally.

I followed this Guide: https://github.com/guino/BazzDoorbell/issues/2 and of cause: https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080

mjpeg and snip is working for my but i want to get rtsp to work

Here is the content of my SDcard:
[sdcard.zip](https://github.com/guino/ppsapp-rtsp/files/6517689/sdcard.zip)

Thank you for your time

guino, 2021-05-20:
@datNoko at a quick look your files appear correct. You said mjped/snap works so your hack is installed correctly and busybox is correct. I even compared your ppsapp with a copy I have of that version and it seems fine based on the version you listed above, so unless your SD card has some problem (i.e. reading/writing issues/corruption), it should be working.

Can you double check if perhaps your ppsapp wasn't updated to a newer version after you did the hack (which could cause problems). You can use the /devices/deviceinfo URL to check it (after removing ppsapp from SD card). Alternatively you can just remove ppsapp **and** the 'home' directory from the SD card and it should re-create the home directory with your current ppsapp in it.

The only other thing I would like to see is the response of your /proc/cmdline -- I doubt that has a problem but just want to double check since I'm not seeing anything wrong right now.

If nothing works you can try a 'hail mary' change which would be to replace these lines in initrun.sh:
```
# Run original app partition code
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStartOrig && chmod +x /tmp/PPStartOrig && /tmp/PPStartOrig &
```
to this:
```
BASE_DIR=/opt/pps
tar xzf ${BASE_DIR}/app.tar.gz -C /
rm -f /home/app/ppsapp
umount ${BASE_DIR}
/home/init.d/initS &
```
Hopefully the above would run the patched ppsapp without running the one from flash first but honestly I have never tested the above (worst case scenario you can restore the original lines and you'll be back where you are now).

datNoko, 2021-05-20:
Thanks for you fast reply,
i removed my home dir and the ppsapp and got a new one with a different MD5 (bb7d90607beeeb45d08fe60f63c02cbe)
I was feeling risky and tried it with the patch that i used before but with the same result (bootloop).
Do i need another patch? 

I want to think it's not the SDcard it was bought 2 days ago.

Here are my Outputs: 
http://192.168.1.116/devices/deviceinfo
`{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"060577727","softwareversion":"2.9.6","hardwareversion":"M11S_H1_V10_F23","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.6.20200628","authkey":"XY4md5wRm8WH693UNBZLMVJpKKWxdfV3","deviceid":"pp016c23d888464d5859","identity":"MR2007010101407670","pid":"aaa","WiFi MAC":"7c:a7:b0:f4:a8:23"}`

http://192.168.1.116/proc/cmdline
`mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino, mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts`

guino, 2021-05-20:
@datNoko can you try copying the ppsapp from home directory into root of SD card (without patching) to see if it works at all ? if the MD5 is different then changes are the file may require different patching -- I don't recommend patching a file if the MD5 doesn't match because highly likely the patch won't work and could cause the device to boot loop too.

You can also try the changes to initrun.sh (without patching) to see if the camera works normally.

datNoko, 2021-05-20:
I have tested if the stock ppsapp boots once in the root, but it doesn't.

When I change the initrun.sh it does nothing, not even beeps.

Now it's getting a bit strange:
I deleted home and ppsapp once again, changed initrun.sh to normal and got the same MD5 as the patched version you posted.
(50ad9c96c65c0e446d8b3d5c8c828957)
but even that doesn't boot once in the root dir.
I then changed the initrun.sh again and it booted
----
I have patched the "new" root bootable version of my ppsapp with MD5 (50ad9c96c65c0e446d8b3d5c8c828957)
and it booted. WITH rtsp working

So i guess recreating the home with the modified initrun.sh was the important part for some reason.

Thanks so much for your help.

For people having the problem in the future here is my working SDcard:
[Working.zip](https://github.com/guino/ppsapp-rtsp/files/6518455/Working.zip)

closed 2021-05-20 by datNoko