guino/ppsapp-rtsp #13: Request - Bell 5T BE5T_H1_V10_433

jonnyrider, 2021-06-25:
Hi,

Can you please patch my psapp for me? Much appreciated!

{"devname":"Smart Home Camera",
"model":"Bell 5T",
"serialno":"xxx",
"softwareversion":"4.3.3",
"hardwareversion":"BE5T_H1_V10_433",
"firmwareversion":"ppstrong-c5-s_meco-4.3.3.20210122",
"identity":"MR2008150100103246",
"uuid":"",
"licence_id":"ppsc2c24404a0b184d7d",
"WiFi MAC":"7c:25:da:5c:db:fd"}

[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6715714/ppsapp.zip)

guino, 2021-06-25:
I'll take a look.

guino, 2021-06-25:
You can try this patch (for onvif) -- this is different from all previous versions so I am not 100% sure onvif will work but mjpeg, snap and play should work:

[ppsapp-onvif.zip](https://github.com/guino/ppsapp-rtsp/files/6717756/ppsapp-onvif.zip) please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x04802b0
play.cgi request address: 0x0483b94

After patching RTSP is available at rtsp://IP:8554/Streaming/Channels/101 and rtsp://IP:8554/Streaming/Channels/102 user admin and password admin. Onvif should be available on port 8000.

NOTE: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, there are also reports that the onvif profile has errors which prevent it from working on certain applications.

**NOTICE**: You should also be sure to try this password for onvif: YWRtaW4=

jonnyrider, 2021-08-22:
> @jonnyrider <https://github.com/jonnyrider> were you able to give this a
> try ? I would like to know so I can add it to the list of patches, thanks.

Apologies, I actually had to return the doorbell because it had an issue
(not related to this) and I haven't been able to check the firmware.

closed 2025-10-01 by jonnyrider