guino/ppsapp-rtsp #14: Unable to find reference init echoshow

josvink66, 2021-07-25:
Hello,

I'm trying to patch my ppsapp file to support RTSP. But i'm unable to locate the "init echoshow" section.
Anyone willing to give it a try for me?

I've attacht the original ppsap file:
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6873892/ppsapp.zip)

Camera details:
"devname":"Smart Home Camera",
   "model":"Mini 16S",
   "serialno":"100149735",
   "softwareversion":"4.0.6",
   "hardwareversion":"M16S_A2_V10_F37",
   "firmwareversion":"ppstrong-a3-tuya2_lsc-4.0.6.20210311",
   "identity":"M1N001CF3800909892",
   "authkey":"SmlyhgEJbdAKRRL3MzDjDmkIFy7Pspxf",
   "deviceid":"pp01f6e88b1f6049db43",
   "pid":"aaa",
   "WiFi MAC":"b4:fb:e3:f5:f6:bf",

Thanks,
Jos Vink

guino, 2021-07-25:
@josvink66 that version of firmware will likely not have the echoshow/RTSP code but it will likely have ONVIF. If you want to patch it yourself have a look at: https://github.com/guino/Merkury1080P#customization otherwise I'll take a look when I have a chance.

josvink66, 2021-07-25:
Hi @guino

Thanks for the reply. Aah that make sense. I'll try to look in more detail in your link to fix it myself.

If you have the chance to look at it, that would be great. I unfortunately have extremely little time because unexpectedly my unborn child (36 weeks) has decided to make its way..

Thanks in advance!
Greetings,
Jos

guino, 2021-07-26:
@josvink66 I posted your patch to the list here: https://github.com/guino/ppsapp-rtsp/issues/1 -- best wishes to you and your family.

josvink66, 2021-07-27:
@guino 

Thank you very much for the patch and the wishes!

josvink66, 2021-08-23:
Hi @guino

Thanks again for the file. I just had some time to play with the camera again. But it seems that not all required ports are open:
[image]
[image]

When I try to connect on my pc with Onvif Device Manager:
[image]

Or on Android:
[image]

So Onvif is detected, it's just unable to connect. Also I cannot connect to the RTSP stream: rtsp://IP:8554//Streaming/Channels/101 Also confirming that port is not open.

Any advise?

Greetings,
Jos

guino, 2021-08-23:
@josvink66 can you post a zip of your SD card files (removing the SDT folder) ? that way i can review and make sure everything is correct. The only thing I can think of is that you may have not patched ppsapp correctly OR you placed the patched ppsapp in the wrong place or with the wrong name (it should be in the root of the SD card named just **ppsapp**)

josvink66, 2021-08-24:
Hi @guino thanks for the reply. This is the content of my SDCard. I've also removed my WiFi credentials in ppsFactoryTool.

[SDCard.zip](https://github.com/guino/ppsapp-rtsp/files/7037914/SDCard.zip)

The ppsapp file on the root of the SDcard is the one you patched for me and placed here; **https://github.com/guino/ppsapp-rtsp/issues/1**

Thanks!

guino, 2021-08-24:
@josvink66 you are missing a bunch of files -- you probably missed this part on #13 after step 7: **Follow the rest of the steps (starting from step 8) from #2**

josvink66, 2021-09-21:
@guino Thanks for the reply. Just had some time again to play with the camera again. You where right on me missing a bunch of files. I've missed a few steps. So I've added the missing files and edited the cgi files, but unfortunately it's still not working. I've run the telnet command that was mentioned here: https://github.com/guino/ppsapp-rtsp/issues/2

`(PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}') kill $PPSID /mnt/mmc01/ppsapp &> /mnt/mmc01/ppsapp.log &)`

The result of that is: `sh: invalid number '/mnt/mmc01/ppsapp'`

I've created a zip file of my SD card again: 
[SDCard.zip](https://github.com/guino/ppsapp-rtsp/files/7203500/SDCard.zip)

Any advise? Thanks!

josvink66, 2021-09-21:
Some extra information. I'm able to connect ONVIF device manager to the camera with authentication. But the stream is not working:

[image]

Port 8554 still seems to be closed:
[image]

snap.cgi and mjpeg.cgi are working!

guino, 2021-09-21:
@josvink66 I don't see this `(PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}') kill $PPSID /mnt/mmc01/ppsapp &> /mnt/mmc01/ppsapp.log &)` in your custom.sh in the SD card zip posted -- please NOTICE if you place a **single line** like that it will not work -- these are supposed to be 3 (THREE) separate commands (one per line), like this:
```
PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
kill $PPSID
/mnt/mmc01/ppsapp &> /mnt/mmc01/ppsapp.log &
```
The above is basically what is already in custom.sh except for the last line which has `&> /mnt/mmc01/ppsapp.log &` at the end.

I am not sure where you got the commands in a single line with parenthesis around but that won't work as posted and that's why you're getting that invalid number message in the log file.

josvink66, 2021-09-22:
@guino thanks for the reply. I got the command from https://github.com/guino/ppsapp-rtsp/issues/2 there it was run in the telnet session. But I indeed see the 3 commands in the custom.sh file. I've changed the ppsapp line to log the output.

The log file is now as follow:
[ppsapp.log](https://github.com/guino/ppsapp-rtsp/files/7208620/ppsapp.log)

guino, 2021-09-22:
@josvink66 the log doesn't show a lot of information, so I'm going to double check the changes in the patch and let you know what I find.

josvink66, 2021-09-22:
@guino great thanks!

guino, 2021-09-24:
@josvink66 you may want to try the information on this post to see if you can get onvif_enable before I have a change to review the patch: https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826 If it works it would save time for both of us.

josvink66, 2021-09-24:
@guino I already try that! Same issues! Sorry should of mention it.

josvink66, 2021-09-26:
@guino Just had some time again. onvif manager is still not working. But the RTSP stream is! I can access it with: rtsp://admin:admin@IP:8554/Streaming/Channels/101 I had an typo in the username/password section... -_-

Portscan still shows port 8554 as closed, but it seems to work now! thanks for all the support. I will continue testing and will report any issues.

guino, 2021-09-26:
@josvink66 glad you got it working - there are a lot of reports the onvif from scratch these devices is buggy so that’s probably why onvif manager isn’t working but if you try different onvif clients you may find one that works.