guino/ppsapp-rtsp #16: ppstrong-a3-tuya2_merkury-4.0.6.20210310

xinterix, 2021-08-02:
I Tried patching  but with no success... Perhaps someone who knows what they're doing can??

devname	"Smart Home Camera"
model	"Mini 11S"
serialno	"100316294"
softwareversion	"4.0.6"
hardwareversion	"M11S_A2_V10_F37"
firmwareversion	"ppstrong-a3-tuya2_merkury-4.0.6.20210310"

guino, 2021-08-03:
@xinterix did you get a ppsapp file that you need patching (in which case you should post it) or what is the problem?

xinterix, 2021-08-03:
file was archived directly off of sd-card

xinterix, 2021-08-03:
[ppsapp.gz](https://github.com/guino/ppsapp-rtsp/files/6924482/ppsapp.gz)
sorry with file here it is...

xinterix, 2021-08-05:
I finally got it patched successfully but I don't know how to make the  .ips file in so i can post it in the  ppsapp file section. 
The patched file is here 
[ppsapp.gz](https://github.com/guino/ppsapp-rtsp/files/6935653/ppsapp.gz)
 and the unpatched one is listed above.

guino, 2021-08-05:
@xinterix sorry I have not had a chance to look at it, but if you got it working: good job!

I can make the patch and post it to the list with credits to you.

guino, 2021-08-05:
@xinterix I posted the patch to the list and added in the mjpeg/snap and play.cgi addresses (don't know if you had/needed those).

closed 2021-08-05 by xinterix

xinterix, 2021-08-05:
Does the port need to be changed in the custom.sh file from 8080 t0 8090 for the newer 4.0* versions? I don't think this was mentioned in the post here: https://github.com/guino/Merkury1080P... The guide found there worked perfect for patching. Just as in the instructions was found three refereces that could write to the onvif value that enables it and all three must patched.

guino, 2021-08-05:
@xinterix short answer: no. Long answer: ppsapp on 4.0.6 uses port 8090 for the URLs but in custom.sh the only port setting is 8080 for httpd which is not related to that at all and should not be changed to any port already in use (or httpd won't work). The purpose of httpd is to allow browsing the files in SD card remotely and allow the cgi files to work (play, snap, mjpeg) -- if you don't use this stuff you don't need to worry about it.

If you want to use snap/mjpeg you must be sure you have these in the SD card:
-the right busybox file (1+ Mb in size)
-the right jpeg-arm file (7.7Kb)
-httpd.conf with a user/password defined
-the cgi files in cgi-bin directory
-the right address setup in snap/mjpeg.cgi

The URL for snap/mjpeg is like this: http://user:password@IP:8080/cgi-bin/snap.cgi    (where user:password is whatever you defined in httpd.conf) -- as an example: if you  change the 8080 in custom.sh to 8888 you should change the snap/mjpeg URL to also be 8888.