guino/ppsapp-rtsp #17: Geenie was able to detect and ban patched devices

xinterix, 2021-08-08:
I went and launched geenie today which I've been using for a while now without any problems with my patched cameras and, perhaps maybe it had an update automatically but when been I opened it up the geenie app today I got a message saying that "jailbroken devices are not allowed" and it then disabled them through the application but it also had the server wasn't working anymore either but until you restarted the device.

xinterix, 2021-08-08:
I posted this in the wrong repository and I don't know how to put it in the correct one

guino, 2021-08-08:
@xinterix this repository is fine but it would be nice to know your device model and firmware version (before&after the issue). I have not heard of anything like this before even on newer devices. By any chance is your **PHONE** jailbroken? 

I use the official tuya app with bazz doorbell and merkury camera (same as geeni). My doorbell currently has a popup asking to update to 2.10.0 but the tuya app lets me choose not to update it automatically so I just cancel out of any update questions.

From your description it sounds like the app locked you out of the device? You should try booting without the SD card (or running the original ppsapp) to see if it works again. In any case I would expect this not to affect how the rtsp/onvif/mjpeg works locally?

Also worth mentioning: if the firmware automatically updated you need to patch the new version and not run the patched ppsapp from the old version as that would have unknown behavior.

xinterix, 2021-08-08:
I was using qemu for Windows 10 on my computer and I had them set up with three cameras one of them was an older 720p version and the other one were two of them they came as a pair and they all came from Walmart I've included an image of them of my screen and how it's was set up at the time. Unfortunately I'm not at home right now and but when I get back I will look at the cuz I know there's going to be an alert in the save the alerts for that the genie app that will show me that same cuz it was came up as alert message which means I'm sure it got logged into the apps log alert section so I'll be able to get that but I'll look more into it when I get back home I know that application was just downloaded and installed from the Play store so it probably was the most recent version but I'll let you know more when I'm back..
[image]

xinterix, 2021-08-08:
i was wrong! It was caused by human error..

closed 2021-08-08 by xinterix