guino/ppsapp-rtsp #19: Psapp patch for B4S_V10_A2_2063 / Firmware 4.0.6

roaringbits, 2021-08-16:
Could you help me patch this ppsapp file?

{
	"devname": "Smart Home Camera",
	"model": "Bullet 4S",
	"serialno": "101020988",
	"softwareversion": "4.0.6",
	"hardwareversion": "B4S_V10_A2_2063",
	"firmwareversion": "ppstrong-a3-tuya2_electro-4.0.6.20210329",
	"identity": "M4D0039H4P02012822",
	"authkey": "ucghm4AU483GfhUyyfpZ9SYkjM3STkHs",
	"deviceid": "pp0179ee6b7175137261",
	"pid": "aaa",
	"WiFi MAC": "84:7a:b6:03:92:16",
	"ETH MAC": "00:00:00:00:00:00"
}

Many thanks! 
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/6992728/ppsapp.zip)

guino, 2021-08-16:
@roaringbits Your patch has been posted to #1

roaringbits, 2021-08-16:
Seems to work great, although I cannot login to the onvif yet (Get a onvif error 400). Any insights? 

Thanks anyways!