guino/ppsapp-rtsp #21: ppstrong-c51-tuya2_geeni-2.9.5.20200324 offline patch request

n8225, 2021-08-23:
Hi @guino, thank you for all of your work on this project.

I did not see an offline patch for this geeni LOOK 1080 cam, could you create an offline patch for it or point me to the correct one to use?

```
{
"devname":"Smart Home Camera",
"model":"Mini 7S",
"serialno":"059992525",
"softwareversion":"2.9.5",
"hardwareversion":"M7S_H1_V11_F23",
"firmwareversion":"ppstrong-c51-tuya2_geeni-2.9.5.20200324",
"authkey":"WCMCIyrITb5fpJuTnwBtmp2iPQkelVgJ",
"deviceid":"pp01f3aefad18647f2d8",
"identity":"MR2004300101401446",
"pid":"aaa",
}
```

Link to ppsapp: https://mega.nz/file/KIRmwb7Q#M5EW4TLZ2QXvf1IZ6D7Aokt1uGTrF0ZR28eLHkDH_qE

The original ppsapp md5 and firmware version match, but the Hardware version differs with the following on https://github.com/guino/ppsapp-rtsp/issues/1:

Firmware Version | Hardware Version | Original ppsapp MD5 | device
-- | -- | -- | --
ppstrong-c51-tuya2_geeni-2.9.5.20200324 | M11S_H1_V10_F23 | 33cd946bc3e4004db60c612133bdd066 | Mini 11S

Is it safe to use this rtsp patch?

guino, 2021-08-23:
@n8225 since version and md5sum are the same: YES - it is safe to use the same patch.

n8225, 2021-08-23:
Is there an offline patch for this version that I missed?

guino, 2021-09-13:
@n8225 sorry for the late reply. I have been playing with the idea that of making a “universal” offline patch. The idea is to let the device boot online and after a minute or so copy a /etc/hosts file blocking the tuya servers so the device doesn’t keep any connections to them.

I have not had a chance to try it yet but this would work on any device/firmware version and would not require setting up/using a ntp server nor would require blocking the internet on the router.

Obviously if you’re savvy enough and have a decent router you can already do something similar in your router with a custom web request.

I will let you know when I have a chance of trying it out.

guino, 2021-09-14:
@n8225 So the 'universal' offline patch works and here are the requirements:
1-The device must have internet access enabled (not unavailable or blocked on router) -- this is so it can connect to the tuya server and enable the RTSP/OVIF features and obtain the current time to set in the device.
2-Create a file in the SD card called `hosts` with the following contents:
```
127.0.0.1 a.tuyaus.com a2.tuyaus.com a3.tuyaus.com m2.tuyaus.com mq.gw.tuyaus.com s.tuyaus.com
127.0.0.1 a.tuyacn.com a2.tuyacn.com a3.tuyacn.com m2.tuyacn.com mq.gw.tuyacn.com s.tuyacn.com
127.0.0.1 a.tuyaeu.com a2.tuyaeu.com a3.tuyaeu.com m2.tuyaeu.com mq.gw.tuyaeu.com s.tuyaeu.com
```
3-Modify your custom.sh to include the section below after `/mnt/mmc01/ppsapp &`:
```
#!/bin/sh
if [ ! -e /tmp/customrun ]; then
 echo custom > /tmp/customrun
 cp /mnt/mmc01/passwd /etc/passwd
 /mnt/mmc01/busybox telnetd
 /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
 if [ -e /mnt/mmc01/ppsapp ]; then
  /mnt/mmc01/set light_onoff 0
  PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  kill $PPSID
  /mnt/mmc01/ppsapp &
```
```
  if [ -e /mnt/mmc01/hosts ]; then
   sleep 180
   cp /mnt/mmc01/hosts /etc
   ifconfig wlan0 down
   sleep 120
   ifconfig wlan0 up
  fi
```
```
 fi
fi
```
The `sleep 180` tells the script to wait 3 minutes (180 seconds) after starting ppsapp so it can connect to the tuya servers, enable RTSP/ONVIF and set the time in the device. The `cp /mnt/mmc01/hosts /etc` copies the hosts file into the system effectively blocking any **new** access to the tuya servers (existing connections are not stopped). The `ifconfig` and `sleep 120` shuts down the wifi for 2 minutes so that any existing connections timeout/drop (making the device go totally offline), then it starts the wifi back up.
Depending on the speed and quality of your wifi the 180 seconds may need to be increased (or can possibly be decreased) -- 3 minutes seems to be enough in most cases for my network. I don't recommend reducing the 120 value or you'll risk keeping open connections to the tuya servers.
If it's not clear: it takes about 5 minutes for the boot process to complete but assuming you're not rebooting the device all the time this should be mostly irrelevant.

Here's the list of connections on my system before the above:
```
tcp        0      0 10.10.10.88:57075       44.230.212.240:443      ESTABLISHED 
tcp        0      0 10.10.10.88:34502       35.164.195.39:8883      ESTABLISHED 
tcp        0      0 10.10.10.88:23          10.10.10.126:43596      ESTABLISHED  (telnet connection)
```
Here's the list of connections after the above changes:
```
tcp        0     62 10.10.10.88:23          10.10.10.126:43596      ESTABLISHED  (telnet connection)
```
The phone app will also be unusable and display the device as 'offline'.

The only 'drawback' from this method is that you can't use it if you have no internet at all. The benefits are:
-No need to additional patching of ppsapp
-No need to create/setup NTP server manually
-No need to configure/block internet access for the device on the network/router

If for some reason you/someone intends to use the device on an isolated network (that has no internet) feel free to request an actual offline patch but if possible (at all) I would like to avoid having to make offline patches as it takes a reasonable time for me to open the ppsapp in ghidra, find the code, make the changes, prepare the patch and post it.