guino/ppsapp-rtsp #24: Pspapp Patch for B2S_V10_H1_2063 2.10.0

Mark051o, 2021-08-31:
Hi Guino

Could you please help me patch the attached, device info is:
{
"devname":"Smart Home Camera",
"model":"Bullet 2S",
"serialno":"061404732",
"softwareversion":"2.10.0",
"hardwareversion":"B2S_V10_H1_2063",
"firmwareversion":"ppstrong-c51-tuya2_akaso-2.10.0.20210803",
"authkey":"kRAoGsHPigPA5eK29Okl4RKMx3iTwyzz",
"deviceid":"pp01ab17a0a3bcd2e086",
"identity":"MR2007200900100008",
"pid":"aaa",
"WiFi MAC":"d4:d2:d6:e9:5f:13"

}
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/7086674/ppsapp.zip)

Thanks!

guino, 2021-09-01:
I am traveling so I won’t be able to patch it until after September 6th

Mark051o, 2021-09-06:
Thanks so much, take your time!

guino, 2021-09-06:
I have posted a patch for your firmware to: https://github.com/guino/ppsapp-rtsp/issues/1