guino/ppsapp-rtsp #25: Patch request for

bafplus, 2021-09-04:
Hi,

Can you please patch my psapp ? Thanks!

{"devname":"Smart Home Camera",
"model":"Mini 14S",
"serialno":"100117202",
"softwareversion":"4.0.4",
"hardwareversion":"M14S_A2_V10_F37",
"firmwareversion":"ppstrong-a3-tuya2_shada-4.0.4.20210115",
"identity":"LCM0029W1L02000816",
"authkey":"x4jzanf8R0k53HAZKLvijtWyWv9mT3eg",
"deviceid":"pp016192b07873afb018",
"pid":"aaa",
"WiFi MAC":"b4:fb:e3:64:02:7a",
"ETH MAC":"00:00:00:00:00:00"}

[My PPA file](https://cockheyt.tplinkdns.com/ppsapp.zip)

bafplus, 2021-09-05:
I have 2 of these camera's.
I dont know if it is of use to you but here's the data for the second (identical) camera:

{"devname":"Smart Home Camera",
"model":"Mini 14S",
"serialno":"100117209",
"softwareversion":"4.0.4",
"hardwareversion":"M14S_A2_V10_F37",
"firmwareversion":"ppstrong-a3-tuya2_shada-4.0.4.20210115",
"identity":"LCM0029W1L02000208",
"authkey":"0LSrxY6VTzhQ5BXlfGwkt9Yer18hRduu",
"deviceid":"pp01ad6f758310163ea3",
"pid":"aaa",
"WiFi MAC":"b4:fb:e3:64:01:c5",
"ETH MAC":"00:00:00:00:00:00"}

[PPA File](https://cockheyt.tplinkdns.com/ppsapp2.zip)

Its for this camera sold under different brands (in this case "Idinio" art.nr: 140101_01 : https://expo.tuya.com/product/703008

guino, 2021-09-06:
@bafplus  They should be the same based on the same output provided - I just got back from a trip so I should be able to patch it soon.

bafplus, 2021-09-06:
That would be great! Thanks in advance!

guino, 2021-09-07:
@bafplus Can you try this patch and report back: 
[ppsapp-onvif.zip](https://github.com/guino/ppsapp-rtsp/files/7122868/ppsapp-onvif.zip)

The code on 4.x firmware changes more often than on 2.x and this variation is new to me so I am expecting this to work but I am not 100% sure. This patch it so enable onvif server on port 8000 and RTSP server on port 8554 with the URLs: rtsp://IP:8554//Streaming/Channels/101 (HD) and rtsp://IP:8554//Streaming/Channels/102 -- please try no user/pass and also try these combinations:
admin:admin
admin:056565099
user: admin and password from /home/cfg/tuya_config.json (onvif_pwd) -if any

bafplus, 2021-09-07:
Thanks, wil do in a few hours.

bafplus, 2021-09-07:
It opened some more ports:

Service
HTTP
Port 23 (TCP)
Port 6668 (TCP)
Port 8000 (TCP)
Port 8080 (TCP)
Port 8090 (TCP)
Port 8554 (TCP)

But still no luck on the rtsp urls using al combo's.

bafplus, 2021-09-07:
Correction!

After a reboot it started to work with username/pass admin:admin

rtsp://admin:admin192.168.1.134:8554/Streaming/Channels/101
rtsp://admin:admin192.168.1.134:8554/Streaming/Channels/102

Many thanks!

guino, 2021-09-07:
@bafplus  Thanks for reporting back I'll add it to the main list.

bafplus, 2021-09-07:
Thanks for the great work!