guino/ppsapp-rtsp #26: Patch request

LeeHeap, 2021-09-20:
Hi

I cannot find an existing patch for my ppsapp, please could the attached be patched? This is from a Bell 5T doorbell running firmware 4.3.3.20210122

Thanks!
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/7195010/ppsapp.zip)

guino, 2021-09-20:
@LeeHeap you need to post the information from /devices/deviceinfo so I can try to put the patch together.

guino, 2021-09-20:
@LeeHeap from yout ppsapp file it looks like this patch may work for it: https://github.com/guino/ppsapp-rtsp/issues/13

LeeHeap, 2021-09-21:
@guino thanks for looking at this. I've tried applying the patch in #13 and the camera seems to boot but will no longer talk to the cloud/mobile app and doesn't run the log parser script for MQTT integration.

My bad forgetting the deviceinfo:
`{"devname":"Smart Home Camera","model":"Bell 5T","serialno":"*********","softwareversion":"4.3.3","hardwareversion":"BE5T_H1_V10_433","firmwareversion":"ppstrong-c5-s_joys-4.3.3.20210122","identity":"MR2008260201202910","uuid":"","licence_id":"ppsc3596823fab4f40ec","WiFi MAC":"7c:25:da:c0:9e:dc"}`

guino, 2021-09-21:
@LeeHeap Please post a zip of your SD card contents without the SDT folder so I can review. The patch should work for your ppsapp so either something is wrong with the patch or there may be something wrong with the SD card files (it could be either but it's easier for me to verify the SD card files first).

LeeHeap, 2021-09-22:
@guino thanks so much for taking a look at this! Here's a zip of my sd card content (credentials have been sanitised)
[sd.zip](https://github.com/guino/ppsapp-rtsp/files/7210324/sd.zip)

guino, 2021-09-22:
@LeeHeap I haven't seen anyone use Dan's files with 4.x successfully yet -- the files may be different enough to cause the issue you're seeing. I would recommend following the instructions here https://github.com/guino/Merkury1080P#conclusion (conclusion section) from the start with a clean SD card and see if your camera can boot normally and allows you telnet access. Once that's working you should try the patch I posted above and report back. I just don't have enough information to know if Dan's files (which are on your SD card) would work so I suggest you don't use them on 4.x firmware.

pioneershahid, 2022-05-18:
Hi @guino, 

hope you are doing well. sorry for hijacking this post. 

As i can see with version of doorbell in this post is same as mine so thought to message here instead of creating new request. Here is info for my doorbell. 
`{"devname":"Smart Home Camera","model":"Bell 5T","serialno":"062800664","softwareversion":"4.3.3","hardwareversion":"BE5T_H1_V10_433","firmwareversion":"ppstrong-c5-s_joys-4.3.3.20210122","identity":"MR2008260201205079","uuid":"","licence_id":"ppscd7a7c5a949204994","WiFi MAC":"7c:25:da:c0:a4:fc"}`

I can enable onvif via cloudedge app. I have frigate installed in HA as addon and recordnig doorbell. I am using `rtsp://192.168.xx.xx:8554/Streaming/Channels/102 `. However the quality of recording is of  SD resolution. the camera have HD and FHD resolution option via app. 

I am not sure if requesting for rtsp patch will allow me two streaming links which i can use for recording?

guino, 2022-05-18:
@pioneershahid did you try rtsp://192.168.xx.xx:8554/Streaming/Channels/101 (**101** instead of 102) ? All cameras that have those RTSP URLS (that I know of) should have both HD and SD resolution available on RTSP.

pioneershahid, 2022-05-18:
@guino I tried it and it didnot work.. i used onzip sniffer and only see 102 in the streaming url.. here is the log for your reference and psapp zip file
[log.docx](https://github.com/guino/ppsapp-rtsp/files/8718857/log.docx)
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/8718863/ppsapp.zip)
.

guino, 2022-05-19:
@pioneershahid I am traveling so I won’t have a chance to look at anything till next week. I have never seen a ppsapp with rtsp on 102 but not on 101 so I am not sure what I would be looking for in the code and have no way of testing without a device, so chances are bleak for any improvement on it.

pioneershahid, 2022-05-28:
@guino just like to follow up if you get chance to look into ppsapp shared thanks

guino, 2022-05-28:
@pioneershahid haven't had a chance to look at it yet.

pioneershahid, 2022-06-06:
@guino  Apologies for the late response.

Please see attached ppsapp log. This is a doorbell using via cloudedge app. I donot think it has PTZ functionality. 
[ppsapp.log](https://github.com/guino/ppsapp-rtsp/files/8848434/ppsapp.log). 

RTSP is only accessable on 102. tried 101 with no luck.  can you please advise where to find tuya_config.json. i cannot seem to find it in the SD Card. thank you.

pioneershahid, 2022-06-06:
@guino another log when i was trying to connect to 101.. i do see rtsp://192.168.1.56:8554/Streaming/Channels/101 path but when trying to access it via motioneye, getting host unreachable message. thanks

guino, 2022-06-08:
@pioneershahid looking at the log: the RTSP server is defintely running on /Streaming/Channels/101 -- I don't really see anything that indicates an error or issue when trying to access but it seems like it is (at some point) trying to access it over a web request (instead of RTSP) -- and that is returning a 401 code which is an error but on a web request (not RTSP request). When using 102 I do not see this web request being made, so the first thing I would check is that you have the same configuration/settings you have for 102 when trying to use 101. The other thing I would try is using VLC, any RTSP or ONVIF phone/pc app, to see if you get the same results (102 working and 101 not working).

If the same settings for 102 don't work on 101 and 101 doesn't work on any other pc/phone app (VLC, etc) then the only explanation would be that the device may be unable to stream the RTSP on main channel due to some sort of hardware limitation and/or permission within the device (and it would be near impossible to find/fix it by patching code).

pioneershahid, 2022-06-08:
@guino  thank you very much. yes it is working and good thing, i can hear audio .. thank you for your help.
appreciate if you can assist with following two queries. 

1. As per `https://github.com/guino/BazzDoorbell/issues/35#issuecomment-991818935`, i got the doorbell to communicate with HA. it was working fine untill SD card corrupted. I had to start over again. I have hack working and but on button press or motion, no mqtt message sent. via telenet, i can see message is coming manually `/mnt/mmc01/mosquitto_pub -h 192.168.x.x -p 1883 -u mqtt -P mqtt -t "homeassistant/binary_sensor/doorbellMotion/state" -m ON`. I saved SD card content [here ](https://1drv.ms/u/s!ArGK6BaouTlKhvhF0MuGxXOIzMkK6g?e=MGegaF) appreciate if you can piont out what i need to fix to send motion/button press mqtt message to HA. 

2.  In custom.sh i enable  removed "#" from ` /mnt/mmc01/set record_enable 0 ` to enable no recording to avoid SD card corruption again. however, i still see video is beng recorded on motion detection in SD card. any pointer here please? thank you

guino, 2022-06-12:
@pioneershahid 
1-In order for the mqtt notifications to work you must have the ppsapp file in the root of the SD card -- even if you don't patch it (it can be the original one), but it must be in there so it can be started with the log being monitored to generate the alerts. On your SD card files I see a ppsapp-custom file but no 'ppsapp', I would just copy the ppsapp file from home/app/ppsapp to the root of the SD card and it should allow you to get the motion and button push alerts. It is possible that having a ppsapp file in the root of the SD card could cause some problems with the video not starting -- if so you'd need specific changes posted with 'Dan's changes' listed here: https://github.com/DanTLehman/orion_sc008ha -- I have only seen 2.9.0 firmware need this but keep that in mind.

2-Some devices may not use/read the setting, so my only advice is to configure the cleanup.cgi file so it can delete files for you regularly so your SD card won't fill up. It is also possible that RTSP may not work correctly on some device if you disable recording entirely.

xromansx, 2022-12-09:
Hi, I have same dorbell, would it be possible to use it with tuya app ?  I would like to incadrate it with alexa but cloudedge app doesn't work good also VLC stream always freezes. The FW in this bells is intercangeble ? I know that CPU and sensor need to be same what about serial nr and other stuff ?

xromansx, 2022-12-10:
It doesn't let me use tuya app the qr code isn't accepted by the cam.I checked OpenIPC, but as you told doorbels not even in the list. I saw there is possibility to change something in FW to use different cloud. But I didn't dig to deep to find out thought maybe someone already done this kind on experiment.

guino, 2022-12-10:
@xromansx the devices on tuya cloud have a unique ID defined in the device and in their servers. If you switched the firmware on a non tuya cloud device for a tuya firmware it would not only have to match the cpu+sensor but would also need to have a valid ID (which can only come from a broken device or from someone who just runs it offline). That’s not something I have done or recommend trying. Probably easier to sell your working doorbell and buy a tuya one. If your device is rooted with RTSP then you can do some integration with it like home assistant, home bridge, domoticz, ifttt, etc but I am not sure what your goal is with alexa, google assist - chances there’s probably a way to do it using the right tools.

guino, 2022-12-10:
Does your cloudedge version have onvif too? Just curious, cause that is a major desirable feature in my opinion.

xromansx, 2022-12-10:
Yes, and from their descpription Cloudedge = onvif , Tuya = automatisation and Alexa/Google.