guino/ppsapp-rtsp #27: Patch request : ppstrong-a3-tuya2_electro-4.0.7.20210624

Janjjr, 2021-10-03:
Hello,
Please any help with patching this '-4.0.7. version. 
There is a already an -4.0.6. version but with differend md5, of course. I did no try this. Should I? 
Anyway, is better to hav the correct patch, i think.

This is the device info:

admin:admin@192.168.0.101:8090/devices/deviceinfo
{"devname":"Smart Home Camera",
"model":"Bullet4S",
"serialno":"100382085",
"softwareversion":"4.0.7",
"hardwareversion":"B4S_V10_A2_2063",
"firmwareversion":"ppstrong-a3-tuya2_electro-4.0.7.20210624",
"identity":"M3J0109H3S02011459",
"authkey":"CteGXtDYl1wrA8lZ88oB5kXRl6IJa4a0",
"deviceid":"pp01f103bcd348bd29e4",
"pid":"aaa",
"WiFi MAC":"84:7a:b6:05:03:6e",
"ETH MAC":"00:00:00:00:00:00"}

And i try to add the 'hacked' ppsapp from the home sub.dir. But i am not shure that it is attached....
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/7273046/ppsapp.zip)

Janjjr, 2021-10-03:
I have indeed tryed the '-4.0.6.' version, It did not work :-(

guino, 2021-10-03:
@Janjjr you should never run a different ppsapp or use a patch for a different ppsapp. I'm looking at your ppsapp now.

guino, 2021-10-03:
@Janjjr there may be a no-patch solution you can try if you like, the steps are here: https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826

I have posted a patch to your version in #1 so you can try anyway.

Janjjr, 2021-10-03:
Hai Wagner,Thanks for the fast respons. I will give it a try.  Verzonden vanuit Mail voor Windows Van: WagnerVerzonden: zondag 3 oktober 2021 18:16Aan: guino/ppsapp-rtspCC: Janjjr; MentionOnderwerp: Re: [guino/ppsapp-rtsp] Patch request : ppstrong-a3-tuya2_electro-4.0.7.20210624 ***@***.*** there may be a no-patch solution you can try if you like, the steps are here: guino/Merkury1080P#9 (comment)I have posted a patch to your version in #1 so you can try anyway.