guino/ppsapp-rtsp #29: no tuya_config.json on camera

mxr0, 2021-11-16:
Hi,

Just a quick question. I followed the rooting guide and have telnet. 
I applied the patch to my ppsapp but i don't see extra open ports.
Upon investigation i don't see a /home/cfg/tuya_config.json
on my camera. From what i see in the custom.sh is that onvif_enable 1 will be set,
But without the tuya_config.json it cannot be changed.
Will this file only be created once connected to the cloud?
Is there a template config i can place on the device, without connecting it to the cloud?

ppsapp md5:
363543651299a6178a4ba77727cc39b5

Deviceinfo:
{"devname":"Smart Home Camera","model":"Mini 16S","serialno":"100584068","softwareversion":"4.0.6","hardwareversion":"M16S_A2_V10_F37","firmwareversion":"ppstrong-a3-tuya2_lsc-4.0.6.20210311","identity":"retracted","authkey":"retracted","deviceid":"retracted","pid":"aaa","WiFi MAC":"00:00:00:00:00:00","ETH MAC":"00:00:00:00:00:00"}

guino, 2021-11-16:
@mxr0 If you don't initialize the device with the tuya app (or manufacturer app) it will remain in a state where most functions will be disabled (including RTSP/ONVIF). I don't know of anything that you can do to initialize it without the tuya app/cloud.

After you initialize it you should have tuya_config.json and the device should enable RTSP/ONVIF with the patched ppsapp (or maybe even with just the json setting). You can at that point remove it from the cloud if you use an offline patch (if available) OR using the universal offline method (which uses the cloud at boot then shuts it off): https://github.com/guino/BazzDoorbell/wiki/Can-I-use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F

sokolWasTaken, 2021-12-21:
> 

Hey, i had the same problem, the file was only created if i initialized the camera with the app, so i made a script that creates the file and fills it with the same settings that the app did, but with onvif_enable set to 1. I did that mainly because i wanted ONVIF to work, but the app overwrites any changes made to tuya_config.json, so i can't set onvif_enable to 1, even when custom.sh is meant to do it.
Thing is, i can't get RTSP or ONVIF to work. Could it be because i didn't initialize it in the app?
Also, is there any way of checking if patched ppsapp is being loaded on boot?

guino, 2021-12-21:
@sokolWasTaken As far as I know the initial app setup of the device does more than just write the tuya_config.json -- I do not believe that creating the json file is enough to get the device in working condition.

If you're using the LSC app you may want to try the tuya app to see if it doesn't cause the device to overwrite the onvif settings every time. You may also be able to 'enable' onvif in the tuya app vs the LSC app -- it is definitely worth a shot.

If you can't get the onvif working with just the json change you may need to patch it - many people have been able to make 4.0.6 work without patching but this could be something related to app, firmware and cloud settings.

With telnet working all you have to do is execute `/proc # ps | grep ppsapp`
If it shows `/mnt/mmc01/ppsapp` then it is running the ppsapp from the SD card (it should be named ppsapp and be in the root of the SD card for it to run). If it shows `./ppsapp` it is running the firmware ppsapp.

sokolWasTaken, 2021-12-23:
@guino 

Hey,
So i managed to get it kinda working, somehow, even with the LSC app, tuya_config finally has onvif_enable 1 with the camera initialized in the app. Patched ppsapp is being loaded on boot too. Thing is, i can connect to RTSP/Onvif, but there's no video. RTSP stream in VLC is just black, there is no connection error or anything like before, with ONVIF, it says "connected" but there is no video, same with the Tuya/LSC app, those just say "retrieving stream".

guino, 2021-12-23:
@sokolWasTaken This is sounding like hardware issues to me. My Merkury 720P camera usually works for many days at a time, then the RTSP/JPEG stops working and the video feed in the app does what you said "retrieving stream" -- I can soft-reboot the device as many times as I want and it won't fix it. It takes usually a couple of POWER CYCLES to fix the problem and then the video feed starts working normally again (for a few hours/days at least). This is something that started happening on this device only and only after several months of normal usage. I have seen this problem on my doorbell and have also noticed the frequency of the problem changes depending on the power adapter I am using. 

Software wise: if it works once chances are it should continue to work unless you change something in the software.
Hardware wise: if will work until something hardware side fails (hardware malfunction, power issues etc).

You could try booting without the SD card a few times to see if the problem happens -- if it does it would tell you that the issue is not related to software but likely something hardware/power related like I mentioned.

If you do suspect of hardware issues you should most definitely try to get it exchanged if/while it is still covered under warranty. If you replace the power supply used and the problem stops then you know the hardware is ok and you just need a better power supply.

sokolWasTaken, 2021-12-23:
@guino 
I'd be guessing it's something with the software, the camera works perfectly fine in the LSC app without the ppsapp patch.
And the RTSP did work in VLC Player and my recorder, not in the app.

guino, 2021-12-23:
@sokolWasTaken as long as you have a way to consistently make it work and you're happy with it -- that's what matters.
Some devices don't even like it when you kill ppsapp and start it again (patched or not). On my Merkury 720 if the hardware is working I can soft reboot or power cycle without issues, but when the issue happens I can soft reboot all I want and it doesn't fix anything -- only after at least one full power cycle is that it comes back.