guino/ppsapp-rtsp #30: Patch request ppstrong-a3-tuya2_general-4.0.7.20210513

Jve91, 2021-12-23:
Hi

Tried to find the snapshot and mjpeg-address myself but couldn't find it.

deviceinfo:

devname | "Smart Home Camera"
model | "Bell 8S"
serialno | "101265497"
softwareversion | "4.0.7"
hardwareversion | "BE8S_A2_V10_433"
firmwareversion | "ppstrong-a3-tuya2_general-4.0.7.20210513"
identity | "M3L0044M5600902552"
authkey | "uY2WKyGMYMApALYnL6Kx5a3moPz8gKz1"
deviceid | "pp01a17fbe0acb907d84"
pid | "aaa"
WiFi MAC | "84:7a:b6:0e:dd:53"
ETH MAC | "00:00:00:00:00:00"

Thanks!
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/7770728/ppsapp.zip)

guino, 2021-12-23:
@Jve91 mjpeg/snap.cgi address: 0395cbc
play.cgi address: 3964cc

I assume you didn't need a patch for it.

Jve91, 2021-12-23:
Wow, fast reply!
I can confirm that it is working now.

No need for a patch indeed.

closed 2021-12-23 by Jve91