guino/ppsapp-rtsp #31: SPEED5S-H1MB_F23 with openipc

ljalves, 2022-01-15:
I have a ppstrong based camera with the hi3518ev300 SoC (board SPEED5S-H1MB_F23) where I have successfully flashed OpenIPC.
No longer on the cloud and I now can stream hls, rtsp, mjpeg, ...
Which SoC does your camera use?

guino, 2022-01-15:
These are the two I have M7C_AK_V10_1245:

[image]

and hi3518ev300 BE8S_H1_V10_915:

[image]

There are many other boards listed here: 
https://github.com/guino/ppsapp-rtsp/issues/1 but we don’t have a lot of chip models because most of the patches were made without having to open the devices.

ljalves, 2022-01-15:
The second one should be able to run OpenIPC (hi3518ev300 soc) - do you know what sensor does it have?

There is a tool to get the hardware info from the camera: https://github.com/OpenIPC/ipctool
Just copy the [binary](https://github.com/OpenIPC/ipctool/releases/download/latest/ipctool) to the sdcard and run it.

ljalves, 2022-01-15:
Btw, I'm gathering all the info about the hacking done on my camera (a SPEED5S-H1MB_F23 board) [here](https://github.com/ljalves/wiki/blob/patch-1/en/device-chacon-ipcam-ri01.md).

guino, 2022-01-15:
> Btw, I'm gathering all the info about the hacking done on my camera (a SPEED5S-H1MB_F23 board) [here](https://github.com/ljalves/wiki/blob/patch-1/en/device-chacon-ipcam-ri01.md).

I was hoping it would take longer for other people to publish that uboot password in order to avoid them from changing it in the future but oh well..

I will try to find some time to run that sensor tool in my device - the hardware id “BE8S_H1_V10_915” indicates the board and sensor but I have no full list of possible values and what they actually represent.

guino, 2022-01-15:
Here's the output for both my devices:
```
/mnt/mmc01 # ./ipctool 
---
chip:
  vendor: SKY39EV2_AK3918E80PIN_MNBD
  model: unknown
ethernet:
  mac: "74:ee:--:--:--:12"
rom:
  - type: nor
    block: 4K
    partitions:
      - name: enc
        size: 0x10000
        sha1: bce80b56
      - name: sys
        size: 0x270000
        sha1: 34476c5d
      - name: app
        size: 0x480000
        sha1: 13051bfa
      - name: cfg
        size: 0xa0000
        path: /home/cfg,jffs2,rw
    size: 7M
ram:
  total: 64M
firmware:
  kernel: "3.4.35 (Mon Dec 7 11:30:43 CST 2020)"
  libc: uClibc 0.9.33.2
  main-app: /mnt/mmc01/ppsapp
```
and:
```
/mnt/mmc01 # ./ipctool 
---
chip:
  vendor: HiSilicon
  model: 3518EV300
  id: 02203c870e0038f1-------------932844f904605c25e3
mdio busy
mdio busy
ethernet:
  mac: "d4:d2:--:--:--:3e"
  u-mdio-phyaddr: 0
  phy-id: 0x00000000
  d-mdio-phyaddr: 0
rom:
  - type: nor
    block: 64K
    partitions:
      - name: enc
        size: 0x10000
        sha1: ef58e856
      - name: sys
        size: 0x310000
        sha1: 96718116
      - name: app
        size: 0x440000
        sha1: 89547352
      - name: cfg
        size: 0x50000
        path: /home/cfg,jffs2,rw
    size: 7M
    addr-mode: 3-byte
ram:
  total: 64M
  media: 27M
firmware:
  kernel: "4.9.37 (Fri Mar 20 21:08:26 PDT 2020)"
  libc: uClibc 0.9.33.2
  sdk: "Hi3516EV200_MPP_V1.0.1.2 B030 Release NoLog (Oct 18 2019, 18:30:38)"
  main-app: /mnt/mmc01/ppsapp
sensors:
- vendor: GalaxyCore
  model: GC2053
  control:
    bus: 0
    type: i2c
    addr: 0x6e
  data:
    type: MIPI
    input-data-type: DATA_TYPE_RAW_10BIT
    lane-id:
    - 0
    - 1
    image: 1920x1080
  clock: 27MHz
```
It is worth mentioning the 2nd one is a doorbell which means 2-way talk and button notifications are a must have (to me anyway).
In the realm of tuya cameras and doorbells these two above are probably the most common hardware out there (along with the [image] which I do not have).

ljalves, 2022-01-15:
Nice! OpenIPC should run fine on the 2nd one but I don't think they have implemented 2way audio yet...
I only see options for streaming in audio an nothing to send:

[image]

It would be interesting to see openipc getting more traction....
I think the original ideia behind the project it is to have something like OpenWRT but for ip cameras.

guino, 2022-01-15:
I would love to see OpenIPC with the support and development level of OpenWRT. 
I just wish I had more time and money to spend buying some devices and tools to work on this stuff -- I started messing with this only because I didn't want to spend a fortune on a doorbell and ended up buying the 2nd camera and a flash programmer with donations I got here, but I don't even own a heat gun to remove the flash chips.

ljalves, 2022-01-15:
By the way, the attached file has the ppsapp of this camera (SPEED5S-H1MB_F23)
A took a quick look but no relevant references to enable onvif:
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/7876014/ppsapp.zip)

guino, 2022-01-15:
@ljalves this is the first 2.8.x firmware I've ever seen, thanks for posting it. It is also the first ppsapp I see that has that /devices/runcmd URL (wish all devices had it -- granted it is a security issue).

At a quick look in ghidra it does seem to have code for some basic onvif discovery (some referenced URLs are /onviif/device_service /search and /devinfo ) -- the URLs seem to be enabled by default but it does go thru information from /home/cfg/pps_appserver.json in order to build up response so you may want to check that file to see if has any useful settings (specifically it looks for an "endpoint" setting).

ljalves, 2022-01-16:
```
/home/cfg # cat pps_appserver.json
{
        "endpoint":     "https://apis-eu-frankfurt.cloudedge360.com",
        "gwCode":       "EU"
}
```

This "ppsapp" is from the original FW shipped with the camera but they already provided an update. I'll get it and post the file again.

ljalves, 2022-01-16:
[newer_ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/7876274/newer_ppsapp.zip)
(might be version 2.9.2)

ljalves, 2022-01-16:
> At a quick look in ghidra it does seem to have code for some basic onvif discovery

The camera does show up when searching the network for onvif devices, but no streaming ports are open! what's the point? right?

I don't understand why manufacturers tend to "cripple" their products - in my opinion they would sell much more if they didn't limit the firmware functionalities...

guino, 2022-01-16:
@ljalves on both ppsapp versions I see the same onvif request handler code but at a closer look it seems they may only be using this as a 'discovery' service and not to provide actual onvif sreaming services. I do see code on both ppsapp versions to provide RTSP function but it doesn't look like there's any setting or code calling the code to start it up (like many versions of ppsapp code I have seen). Is there RTSP working at all on the device you have (or may be an app option to enable it) ? I can try to make a patched ppsapp with RTSP enabled if you'd like to try it -- I don't suppose you need it but I figured I would ask.

guino, 2022-01-16:
I also see mjpeg/snap.cgi address 0x0357d54 and play.cgi address 0x0358770 which should work with the cgi-bin scripts from https://github.com/guino/BazzDoorbell/tree/master/mmc using jpeg-arm and busybox from the SD card (similar to other tuya devices) -- which should work even without patching ppsapp (you would need to start up busybox httpd on the device by hand or add it to some boot script).