guino/ppsapp-rtsp #32: Patch Request

Rastastical, 2022-01-16:
Hi! I hope you're still doing these, thank you very much.
I wasn't able to find an existing patch for it.

{"devname":"Smart Home Camera",
"model":"Bullet 4S",
"serialno":"103112647",
"softwareversion":"4.0.7",
"hardwareversion":"B4S_V10_A2_2063",
"firmwareversion":"ppstrong-a3-tuya2_merkury-4.0.7.20210624",
"identity":"M5H0025C6Q01411685",
"authkey":"nUC49hnET0tqaNHK0zTADu6ZaFyG8DZN",
"deviceid":"pp014ff10e93bcfd21c8",
"pid":"aaa",
"WiFi MAC":"84:7a:b6:ed:fe:d6",
"ETH MAC":"00:00:00:00:00:00"
[ppsapp-zipped.zip](https://github.com/guino/ppsapp-rtsp/files/7876540/ppsapp-zipped.zip)
}

Rastastical, 2022-01-16:
should also add that i do not have a home/cfg/tuya_config even though i set up the camera with the app and everything.

guino, 2022-01-16:
@Rastastical please notice the path to tuya_config.json is /home/cfg/tuya_config.json (notice the first **/** ) — it is NOT on the SD card (it is in the flash memory) so you have to telnet into the device and either copy it to the SD card, edit it and copy it back to flash OR you can edit it in the flash directly using **vi** which is harder to do.

Someone wrote a few lines for initrun.sh that you could use to get it copied to the SD card and back:

Modify initrun.sh:
Add above the `while` line

```
if [ ! -e /mnt/mmc01/tuya_config_original.json ]; then cp /home/cfg/tuya_config.json /mnt/mmc01/tuya_config_original.json; fi
```

Add after the ` sleep 10 ` line:

```
if [ -e /mnt/mmc01/tuya_config.json ]; then
 cp /mnt/mmc01/tuya_config.json /home/cfg/tuya_config.json;
fi
```

Boot the doorbell once, it will create the tuya_config_original.json file in the root of SD card, copy it and rename it to tuya_config.json, open it and set onvif_enable to 1 and adjust any other settings. On the next boot the edited file will copy the json file to the flash.

Please check/try the above first.