guino/ppsapp-rtsp #33: Get snap/mjpeg/play address

AstroxV, 2022-01-31:
I patched my LSC Indoor camera and have everything working without patching pssapp but just enabling stmp in the config.
{"devname":"Smart Home Camera","model":"Mini 16S","serialno":"103781031","softwareversion":"5.2.4","hardwareversion":"M16S_A2_V10_MIS","firmwareversion":"ppstrong-a3-tuya2_lsc-5.2.4.20211015","identity":"M960049J9R01208245","authkey":"b5xPJr7CXV6bS7A2u40JCXdemNkpgq02","deviceid":"pp01927a06f52ba9d7a9","pid":"aaa","WiFi MAC":"f4:b1:9c:16:66:74","ETH MAC":"00:00:00:00:00:00"}

I'm wondering how I can find the address I have to use for the cgi-bin files. 
The firmware is not listed, is there anyway I can figure out the address?
The guide mentions ghidra but I'm not familiar with it. I will look more into it, but I have no idea what to look for.

guino, 2022-01-31:
@AstroxV first 5.x firmware I hear confirmation of working with the hack. 

You can post a zip of your ppsapp and I can look up the jpeg address for you (if available).

The guide I had posted was for 2.x so I doubt it will be similar for 5.x

AstroxV, 2022-01-31:
Thank you. Yeah I'm going through the guide just learning a bit but it might be a bit too complicated for me. I attached the ppsapp, I hope you can find something from it. If you need anything else, please let me know. Already appreciate the work, RTSP from these cheap camera's is great :)

[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/7973829/ppsapp.zip)

guino, 2022-01-31:
This was surprisingly similar to previous versions:
mjpeg/snap.cgi address: 0x0305ba4
play.cgi address: 0x03069a8

Let me know if that doesn't work.

AstroxV, 2022-01-31:
Thank you so much, it's working beautifully. I kinda wish the mjpeg stream had a better framerate, so might setup some server to restream the rtsp, but it is what it is. I tested the play.cgi and also works.
This is great :) I'm gonna keep an eye on your work, really like reading about your exploration on the hardware side. 
Keep up the good work and have a few beers on me.

guino, 2022-01-31:
@AstroxV Thanks for the beers and for reporting back! I will add the addresses and firmware info to the main list so others can find it in the future. 

Unfortunately there's no easy way to improve the mjpeg frame rate (other than using rtsp2jpeg -- which I do have an [existing project](https://github.com/guino/rtsp2jpeg/blob/main/ezvizmjpeg) for it using ffmpeg/scripts). The jpeg buffer on the device is filled up hardware side so it takes scanning the buffer multiple times to find a 'clean' image to stream and that is somewhat slow on these low powered CPUs. 

Another project I've been meaning to work on is to 'add' ONVIF support for devices that only support RTSP but it seems to be more a fading idea since most newer devices support onvif already.

Enjoy and remember to backup your files and/or configure cleanup.cgi so you don't get the SD card formatted by surprise!