guino/ppsapp-rtsp #34: Mini 16S - snap mjpeg OK but not rstp / onvif

valvin1, 2022-02-06:
Hello,

I'm running a Mini 16S bought as a *LSC Smart Connect*.

It is a device running firmware 4.x on which there are some differences like no http port at startup and requires `ppsFactoryTool.txt` which enables `8090` port.

Here is the device info:

```
{"devname":"Smart Home Camera","model":"Mini 16S","serialno":"101***","softwareversion":"4.0.6","hardwareversion":"M16SA2V10F37","firmwareversion":"ppstrong-a3-tuya2lsc-4.0.6.20210311","identity":"M4L007CF5S00****","authkey":"****","deviceid":"***","pid":"aaa","WiFi MAC":"84:7a:b6****","ETH MAC":"00:00:00:00:00:00"}
```

At the moment of this ticket I've succeed in having telnet and httpd working (only snap.cgi and mjpeg.cgi).

I did the `ppsapp` patch using the recommended website with initial and path files. Initial md5sum file is corresponding to the one I get in `/home/app`

But and this the reason of this issue rstp and onvif are not working. There are no other ports opened than the tuya port, httpd, telnet and the 8090 port we activated at the first step.

confirmed by using netstat:

```
netstat -lt
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 0.0.0.0:6668            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:webcache        0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:telnet          0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:8090            0.0.0.0:*               LISTEN  
```

using a tool like `Onvif Viever` my camera is detected on port `8000` but it doesn't work as expected because nothing listen on this port.

concerning rstp I tried the different url on port `8554` but same as onvif no port are listening.

my camera has internet connection and can contact the internet.

i tried to get the log of the patched `pspapp` but nothing relevant appears.

Do you have any idea on what I did wrong?

I'm wondering if the fact I associated the camera with LSC app one time is not doing a lock thing?

guino, 2022-02-06:
@valvin1 the device must be setup with the app (lsc, tuya, smarlife — whatever app you prefer), so it will connect to tuya servers before it can start rtsp/onvif functions.

Most 4.x versions work without patching if you follow this post: https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826

if you have tried the patch and the config changes and it still doesn’t work post a zip of your sd card without the SDT folder and I can check it. A lot of the times people place the patched ppsapp on the wrong place or with the wrong name (it should be called **ppsapp** and be on the root of the SD card for it to work. 

if you managed to make a log (even if small) please include it in the zip file if you end up posting one.

guino, 2022-02-06:
I noticed now the only patch we have for your version is the offline patch — this is only meant to be used without connecting the device to the internet, but please notice you still need to have the device enrolled on an app (tuya, lsc, smartlife, etc) so the device is not in ‘factory mode’ which could prevent it from starting up rtsp/onvif.

valvin1, 2022-02-06:
thank you @guino I think the error is I had reset the device and never enrolled it again. I believed it was better to do like that.

I think the ppsapp is placed correctly if I refer to the `custom.sh` file and `ps` command:

```
  426 valvin     972 S    {initrun.sh} /bin/sh /mnt/mmc01/initrun.sh
...
  735 valvin    1252 S    /mnt/mmc01/busybox telnetd
  741 valvin    1252 S    /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
  747 valvin    116m S    /mnt/mmc01/ppsapp
  872 valvin    1908 S    wpa_supplicant -B -Dnl80211 -iwlan0 -c/etc/wpa_supplicant.conf -C/var/run/wpa_supplicant -P /var/run/wpa_suppl
  927 valvin     984 S    udhcpc -b -i wlan0 -s /etc/udhcpc.script -p /var/run/udhcpc-wlan0.pid -x hostname 101604522
 1412 valvin    1260 S    /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
 1413 valvin     976 S    {mjpeg.cgi} /bin/sh mjpeg.cgi
 1419 valvin     752 S    /mnt/mmc01/jpeg-arm /proc/747/mem 037c2f4 57336 mjpeg
 ...
```

By the way thanks `cgi-bin/mjpeg.cgi` I get motion detection almost working :)

I'll try the following:

* enroll the cam on the app again without the sd card
* remove internet connection to the app (which sounds good for my privacy :) )

If it doesn't work i'll grab logs and the content of the sd card in a zip file.

valvin1, 2022-02-12:
After adding my camera using the corresponding app Onvif and rstp are working as you mentioned.

On the device i'm using it is not an offline patch it looks there are two different Mini 16S model.

This morning I also bought a Bullet 4S (outdoor). I repeat same procedure and get everything working. I can have mjpeg, onvif and rtsp working. But it requires camera having an internet connection as my Mini 16S to have onvif and rstp (it works in all case for mjpeg).

The md5sum of my ppsapp doesn't match with the offline Mini16S version. Is there a specific procedure to have an offline patch ? It looks offline firmware on Mini16S has been done by @jilleb ?

valvin1, 2022-02-12:
unfortunately hosts file doesn't work out of the box but with this error:

```
call user###pps_media_md_task_loop_call_line:1490
crc:e49c
[00:00:29.460 INFO pps_tuya_netmng.c:678]WIFI[0] Get IP:192.168.1.124
[00:00:31.050 INFO pps_device_led_ctrl.c:179]led event: net connected
store dns info to position 7 for m2.tuyaeu.com 0x7f000001
[01-01 01:00:27:366 TUYA Debug][mqtt_client.c:1145] select mqtt host:m2.tuyaeu.com
[01-01 01:00:27:366 TUYA Notice][mqtt_client.c:1158] mqtt get serve ip success
[01-01 01:00:27:392 TUYA Debug][mqtt_client.c:822] serverIP 2130706433 port:8883
[01-01 01:00:27:393 TUYA Err][mqtt_client.c:1174] mqtt socket connect fail. retry_cnt:3 op_ret:-703 errno:-25
[01-01 01:00:28:403 TUYA Notice][mqtt_client.c:1326] MQ_EXIT_ERR : mqtt close
[01-01 01:00:28:403 TUYA Debug][mqtt_client.c:701] close mqtt
[01-01 01:00:28:403 TUYA Debug][mqtt_client.c:1345] mqtt fail_cnt:4 and sleeptime:2915 ms
```

it might be because it can't join the mqtt server. I see there was some binaries for that is there extra step to make it work?

guino, 2022-02-12:
@valvin1 if you block the internet to the device (by firewall or by hosts file) the device will jot start up enough to enable RTSP/ONVIF unless you use an offline patch (which has to be made for the specific firmware). If you have not tried it, this would be an option to only have the device online enough to startup then disable internet to it:
https://github.com/guino/BazzDoorbell/wiki/Can-I-use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F

if the above doesn’t work I can try to make an offline patch for it, but please confirm you need the offline patch for ppstrong-a3-tuya2lsc-4.0.6.20210311 so I don’t spend time on the wrong firmware. If you can use the offline without patching (link above) it does save me time from digging into the firmware, plus I have no way to guarantee it will work since I don’t have the device.