guino/ppsapp-rtsp #37: Please help patch for Merkury 1080p 4.0

RossCobb, 2022-03-12:
Wonderful work on this everyone.  Confusing as heck for a github newbie though.
I've gotten to various stages including telnet and viewing the hack "done" but never could find or generate the json file that needed editing. I'm guessing that this technique requires the ppsapp to be patched first?

Thanks for your help and patience.  Small donation sent.  
Ross

Firmware 4.06 as updated from Geenie app today.

CRC32:
87a8ad87
MD5:
db7cda9a70bcb1e7c5343dce00544eba
SHA-1:
5b8d683e917f03726c875f80efaf43f2e108c47a

[ppsapp-merkury-1080p-406.zip](https://github.com/guino/ppsapp-rtsp/files/8238715/ppsapp-merkury-1080p-406.zip)

guino, 2022-03-13:
@RossCobb if you got telnet working you can copy the json file to the sd card with this telnet command: 
`cp /home/cfg/tuya_config.json /mnt/mmc01`
then wait a few seconds (ie 30s) and edit the file in the SD card by accessing the SD card on a computer. After editing you can reboot up with the SD card in the device again, Telnet into it and use this command to copy the json file into the device:
`cp /mnt/mmc01/tuya_config.json /home/cfg` then wait a few seconds (ie 30s) and reboot the device so the modified json file is used.

if you haven’t got telnet working yet, please zip/post the contents of your SD card without the SDT folder so I can review it.

By the way: thanks for the coffees!

RossCobb, 2022-03-13:
You're quite welcome!

For the life of me I've never seen that or any json file generated. I've
browsed that folder and many others.

I recall reading somewhere that it was generated when activating the
camera... But when I boot from SD it disappears from the app.
I'm just lost ...lol

guino, 2022-03-13:
@RossCobb if you need to send attachments please reply in github as replying by email does not send any attachments (or use my direct email address on my github profile).