guino/ppsapp-rtsp #38: Snap for Mini 11S (Firmware 4.0.7.20210624, Merkury 1080p) - No "pps_media_md_get_pic" Found

richaardvark, 2022-05-24:
Hello,

Trying to get snap and mjpeg working on my 4.0.7 firmware version 1080p Mercury camera.  Searching the code in Ghidra for "pps_media_md_get_pic" returns no results :(  What do I do next??

I already have rtsp working.

```
   "devname":"Smart Home Camera",
   "model":"Mini 11S",
   "serialno":"102783758",
   "softwareversion":"4.0.7",
   "hardwareversion":"M11S_A2_V20_GC1",
   "firmwareversion":"ppstrong-a3-tuya2_merkury-4.0.7.20210624",
```

Thanks!

richaardvark, 2022-05-25:
mjpeg.cgi still isn't working. The URL loads but all I see is a tiny grey square symbol and no image.

richaardvark, 2022-05-26:
**For snap:** I followed the [directions here](https://github.com/guino/ppsapp-rtsp) closely but there was no "pps_media_md_get_pic" found even though my code/scripts looked almost exactly like your screenshots in the guide.

*For mjpeg:** I followed the directions closely and found the address: 0x003940e4. I updated the mjpeg file with that address and the URL will load but there is just an empty grey square displayed. I'm really hoping you will see this message soon and have a moment to help me. Thank you!

guino, 2022-05-26:
@richaardvark the 4.0.7 firmware has several differences from the posted information -- the differences can be big even between two different 4.0.7 firmware files. Please post a zip of your ppsapp and I can find the address for you.

richaardvark, 2022-05-26:
> @richaardvark the 4.0.7 firmware has several differences from the posted information -- the differences can be big even between two different 4.0.7 firmware files. Please post a zip of your ppsapp and I can find the address for you.

Here is a zip with my ppsapp.  Thank you very much for your help!
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/8781907/ppsapp.zip)

guino, 2022-05-27:
@richaardvark your snap/mjpeg address should be: 0x03932e8

richaardvark, 2022-05-27:
> @richaardvark your snap/mjpeg address should be: 0x03932e8

Thanks! So I edit both of the cgi files with that address and save and it should work, right? I don't need to change any other files?

richaardvark, 2022-05-27:
> @richaardvark your snap/mjpeg address should be: 0x03932e8

Thanks!  I updated both my snap.cgi and mjpeg.cgi files with that address but neither URL will load.  It never times out but just keep indefinitely loading forever:

```
#!/bin/sh
# JPEG address (only enter the digits after 0x)
ADDRESS=03932e8
# Get ppsapp PID
PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
/mnt/mmc01/jpeg-arm /proc/$PPSID/mem $ADDRESS 57336 2> /dev/null
```

[http://admin:admin@192.168.1.242:8090/cgi-bin/snap.cgi](http://admin:admin@192.168.1.242:8090/cgi-bin/snap.cgi):
[image]

guino, 2022-05-27:
@richaardvark I will double check the address on the code and let you know, meanwhile please check:
* you should NOT have ppsFactoryTool.txt file on the SD card as this may prevent the jpeg function from starting
* since you did not patch ppsapp you should not have ppsapp in the root of the SD card
* the port for using snap/mjpeg on your screenshot is wrong — you need to use the port defined in custom.sh in the httpd line (usually 8080) along with the user and password configured in httpd.conf (try browsing the SD card with just http://user:password@102.168.1.242:8080/ with the user and password from httpd.conf and if it lists the SD card files it should be working), notice that if you modify custom.sh or httpd.conf you have to reboot the device for changes to take effect.
* make sure you have the support files in the SD card (jpeg-arm, httpd.conf, cgi-bin/main.cgi, etc)

richaardvark, 2022-05-27:
> * you should NOT have ppsFactoryTool.txt file on the SD card as this may prevent the jpeg function from starting

Oops - I *did* have the ppsFactoryTool.txt file on the SD card, in the root folder.  It is now deleted.  I thought that according to [these instructions](https://github.com/guino/Merkury1080P#conclusion) though we needed the ppsFactoryTool.txt file and that it is necessary for enabling http server URLs, which are at port 8090 for the 1080p camera, not 8080? 🤔

> * since you did not patch ppsapp you should not have ppsapp in the root of the SD card

I had manually added it but it is now removed.

> * the port for using snap/mjpeg on your screenshot is wrong — you need to use the port defined in custom.sh in the httpd line (usually 8080) along with the user and password configured in httpd.conf [...]

Yes, navigating to the URL with port 8080 (matches custom.sh) does list all the SD card files, so that's a good sign.

> * make sure you have the support files in the SD card (jpeg-arm, httpd.conf, cgi-bin/main.cgi, etc)

Here are all files now listed on my SD card/in the cgi-bin folder:

```
/mnt/mmc01 # ls
upload.html                index.html                 bin
set                        httpd.conf                 etc
ppsMmcTool.txt             hosts                      lib
passwd                     env                        home
offline.sh                 custom.sh                  System Volume Information
mqtt_pub                   busybox                    sdrec
jpeg-arm                   cgi-bin
initrun.sh                 hack
/mnt/mmc01 # ls /mnt/mmc01/cgi-bin
upload.cgi   snap.cgi     mjpeg.cgi    main.cgi     cleanup.cgi
/mnt/mmc01 #
```

I deleted the ppsFactoryTool.txt file and the ppsapp file from the root of the SD card and confirmed everything else/all addresses are correct and am rebooting now and will see what happens... fingers crossed 🤞🏻

guino, 2022-05-27:
@richaardvark glad snap/mjpeg and rtsp are working. 

Regarding the URLs on port 8090 these will only work with ppsFactoryTool.txt in the SD card. We only need one these URLs to get information to install the hack but if you’re using it for other purposes you can try keeping it in place but if RTSP/snap/mjpeg stop working after a reboot of the device you should keep in mind the ppsFactoryTool file be the reason. You can add the ppsFactoryTool anytime after boot and it should enable the URLs but some firmware versions will not initialize the video (rtsp/snap/mjpeg) if that file exists during boot — if your version is like that you may need to do something different to create the file later after the device has booted up so you can use the URLs.

richaardvark, 2022-05-27:
> @richaardvark glad snap/mjpeg and rtsp are working.
> 
> Regarding the URLs on port 8090 these will only work with ppsFactoryTool.txt in the SD card. We only need these URLs to get information to install the hack but if you’re using it for other purposes you can try keeping it in place but if RTSP/snap/mjpeg stop working after a reboot of the device you should keep in mind the ppsFactoryTool file may be the reason. You can add the ppsFactoryTool anytime after boot and it should enable the URLs but some firmware versions will not initialize the video (rtsp/snap/mjpeg) if that file exists during boot — if your version is like that you may need to do something different to create the file later after the device has booted up so you can use the URLs.

ok, thanks! :)  Leaving the ppsFactoryTool file there doesn't seem to be breaking anything or preventing the jpeg stream from starting.  If it does perhaps I can write some script somewhere to add the file upon reboot but delayed? 🤔

guino, 2022-05-27:
> > @richaardvark glad snap/mjpeg and rtsp are working.
> > Regarding the URLs on port 8090 these will only work with ppsFactoryTool.txt in the SD card. We only need these URLs to get information to install the hack but if you’re using it for other purposes you can try keeping it in place but if RTSP/snap/mjpeg stop working after a reboot of the device you should keep in mind the ppsFactoryTool file may be the reason. You can add the ppsFactoryTool anytime after boot and it should enable the URLs but some firmware versions will not initialize the video (rtsp/snap/mjpeg) if that file exists during boot — if your version is like that you may need to do something different to create the file later after the device has booted up so you can use the URLs.
> 
> ok, thanks! :) Leaving the ppsFactoryTool file there doesn't seem to be breaking anything or preventing the jpeg stream from starting. If it does perhaps I can write some script somewhere to add the file upon reboot but delayed? 🤔
> 
> I have been using some of the [URLs from this list here](https://github.com/guino/BazzDoorbell/issues/49) and have found them helpful for controlling the camera without using the Android app.