guino/ppsapp-rtsp #40: Just wanted to see if a fix for this issue has been found.-

guino, 2022-07-06:
Just wanted to see if a fix for this issue has been found.-
Above already exists a patch for "ppstrong-c51-tuya2_geeni-2.10.1.20210207"
Device info-{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"060343649","softwareversion":"2.10.1","hardwareversion":"M11S_H1_V10_F23","firmwareversion":"ppstrong-c51-tuya2_geeni-2.10.1.20210207","authkey":"lozPNpSAnbJwVhz7GKp4iNV5l5UpW9I9","deviceid":"pp0132b1202c630af158","identity":"MR2005150400913973","pid":"aaa","WiFi MAC":"38:01:46:1e:4d:b3"}

ZIP file is provided above but like the post says Some are reporting rtsp and onvif still not available and this is my issue. (RTSP is needed)
I repatched my ppa with the Rom site and provided zip.
The hack reported "done" with no problem though so I was happy on this :)
I would appreciate any fix or info you could share.

Thank you for your hard work on this whole project it is very much appreciated and I have learned a lot.

_Originally posted by @SoloAsh in https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-1175878582_

guino, 2022-07-06:
@SoloAsh I am not sure what you're trying to 'fix'. If the patch is in the list it's been verified as a working patch by someone with the same exact firmware so it should work for all devices in that same exact firmware. 

The patch file listed is: https://github.com/guino/ppsapp-rtsp/files/6135730/ppsapp-onvif.zip and if you followed the how to patch steps from https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080 then you should have the RTSP feed available at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 with user **admin** and password **admin**. Onvif should be available on port 8000.

If something isn't working you need to provide more details on what you have tried, including what phone/pc applications were tried, user/passwords, error messages received, etc so we can try to help you.

guino, 2022-07-10:
@SoloAsh Do you hear two startup sounds when the device boots up ?

The device is definitely rooted (good), and as long as you hear two startup sounds it means it is loading the patched ppsapp.

Things you need to check:
1-I'm pretty sure web browser (including chrome) are NOT able to view RTSP streams directly, you would at least need some plugin so I recommend you NOT use any web browsers for testing RTSP.
2-In VLC the ONVIF RTSP URLs you're using need to have user:password but you're using the wrong password for them -- the user:password should be admin:admin like: `rtsp://admin:admin@194.168.1.4:8554//Streaming/Channels/101`
3-In VLC the URL for non-ONVIF RTSP stream (which is enabled by the patch) should be on port **8555** with no password, basically: `rtsp://192.168.1.4:8555`
4-For ONVIF clients you need to provide a user:password for it to work, which again it should be admin:admin -- again the reason for providing the patch (for RTSP on port 8555) is because the ONVIF+RTSP implementation is faulty and doesn't work with many clients so even with the correct user:password it may not work (some ONVIF phone apps seem to work best with it).
5-Some cameras have a setting in tuya_config.json which defines the ONVIF password -- the easiest way to check this on your camera is to open the URL: http://admin:056565099@192.168.1.4/proc/self/root/home/cfg/tuya_config.json and see if there's an ONVIF password defined -- if there's a password defined for ONVIF that is what you should use with the ONVIF style URLs in VLC, ONVIF client etc.

SoloAsh, 2022-07-11:
Thank you for looking over this with me. 
so the cam is only making 1 Start up the sound I do not believe it is loading the patched ppsapp.
When I go to provided link to check tuya_config.json it says "This page isn't working". 
I checked Vlc again with provided link making sure to put admin:admin.
Also, check onvif with 2 mobile apps. Onvier.
In my other Onvif viewer I added the admin:admin credential without any luck.
After plugging the cam back Ip changed to 192.168.1.13 (DHCP). I reserved the IP this time. I was
really careful to put in new  IP with the credentials while checking the links.  I was able to bring up "http://192.168.1.13/devices/deviceinfo" .So it seemed the ip change did not hurt anything.
Thank you again.

guino, 2022-07-14:
@SoloAsh post a zip of your SD Card contents without the SDT folder so I can review it.

SoloAsh, 2022-07-15:
Here is a zip of the card. I looked for SDT folder but did not find it. 
Thanks again for the help.
[sd.zip](https://github.com/guino/ppsapp-rtsp/files/9123835/sd.zip)

guino, 2022-07-21:
@SoloAsh You patched ppsapp but your ppsapp file is saved as 'ppsapp.txt' in the SD card (likely web browser put that name when saving it). Please rename ppsapp.txt to ppsapp on the root of the SD card and reboot the device -- you'll likely hear 2 startup sounds then you can try RTSP again with the information I posted here: https://github.com/guino/ppsapp-rtsp/issues/40#issuecomment-1179764636

SoloAsh, 2022-07-27:
It seems it adds the .txt to the end when I zip it. I think because it is a .txt file? I opened with notepad and tried to save as "all files" with the name being just "ppsapp". What should the file type be and how could I convert it? 
Sorry took me a bit to reply. Things got busy
Thanks for your help on this.

SoloAsh, 2022-07-27:
I repatched and was able to download the new file without TXT extension. I am getting 2 startup tones now.
I am still having issues viewing RTSP and onvif.  Same exact issues as before 
Tried the same programs and links as before
Used links with admin, admin, and without the credentials. Onvif is recognized but cannot view it either same as before.

I am getting 2 startup tones now. so i am a step closer.

guino, 2022-07-27:
@SoloAsh can you post the error message you get on VLC when using URL **rtsp://192.168.1.4:8555** ?