guino/ppsapp-rtsp #41: patch request Action LSC Smart Connect video-deurbel

MonkeyMan007, 2022-08-05:
there is a patch but the MD5 don't match up

[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/9270949/ppsapp.zip)

{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"060876946","softwareversion":"2.10.5","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.10.5.20210806","authkey":"MkU6H6klYAxBXjOudpAoMpG5AqrQy96W","deviceid":"pp013e17a8e01451daed","identity":"MR2007120100112794","pid":"aaa","WiFi MAC":"d4:d2:d6:b1:f0:38"}

mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,T mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppPart

#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED="[1;31m"
NORMAL="[0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
#MTDNUM=5

echo "------------->mtdnum:${MTDNUM}"

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7|8)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)     
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart && chmod +x /tmp/PPStart && /tmp/PPStart

guino, 2022-08-05:
@MonkeyMan007  Your ppsapp file is the same as posted in the list (md5 is fa879a12c0194222afb9d3d0e80dafea) -- this is your patch:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_lcs-2.10.5.20210806 | BE8S_H1_V10_433 | fa879a12c0194222afb9d3d0e80dafea | Bell 8S |

NOTE: This version should have RTSP available without patching at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 -- see: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-745618597

NOTE2: Some people reported issues with ffmpeg not being able to play RTSP at the above URLs, and in that case you can try patching with the patch below:

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/7118695/ppsapp-rtsp.zip)  use port **8555** (rtsp://IP:8555) as ONVIF may already be using port 8554, please use this [HOW TO PATCH GUIDE](https://github.com/guino/ppsapp-rtsp/issues/1#issue-759079080)
snap.cgi and mjpeg.cgi address: 0x056e6b0
play.cgi address: 0x0571f94

MonkeyMan007, 2022-08-05:
any way i can get the contacts on the back to switch?

guino, 2022-08-05:
@MonkeyMan007 you mean the same as when you press the doorbell button?

MonkeyMan007, 2022-08-05:
yes

guino, 2022-08-05:
@MonkeyMan007 that is an odd request. Care to elaborate why?

The only reason I can think for this would be for something else to ring the bell (ie separate button) — all you need to do that is wire up another button in parallel and activate it with a second switch or whatever was going to trigger the bell contact like you want it to do by request. 

I honestly have never looked into the code to see what it would take to do that (or to know if it is possible).

MonkeyMan007, 2022-08-05:
no i want to hook up a Hardware bell to my smart video doorbell. so when the button is pressed on the smart doorbell a chime wil go off in mij house

guino, 2022-08-05:
@MonkeyMan007 the doorbell already does that function, meaning: when you press the button it triggers the back contact so your chime rings - the only requirement is that it has to be a mechanical chime and you have to wire it up as described in the smart doorbell manual (most require an adapter to be placed on the chime). There are also some wireless chimes (433mhz) that work with the lsc smart doorbells. You can double check in the app that your doorbell setting is set to “mechanical” (it is an option on the tuya app at least).

MonkeyMan007, 2022-08-06:
I have tested this but it does not do the things u say. sadly I am using the Tuya app not the lcs one

MonkeyMan007, 2022-08-06:
"version":      1,
        "sleep_mode":   0,
        "alarm_fun_onoff":      0,
        "alarm_fun_sensitivity":        0,
        "alarm_fun_mode_switch":        0,
        "alarm_fun_time_start": 0,
        "alarm_fun_time_end":   0,
        "flip_onoff":   0,
        "light_onoff":  1,
        "night_mode":   0,
        "sound_detect_onoff":   0,
        "sound_detect_sensitivity":     0,
        "watermark_onoff":      1,
        "event_record_time":    60,
        "enable_event_record":  2,
        "record_enable":        0,
        "motion_trace": 1,
        "motion_area_switch":   0,
        "motion_area":  "",
        "motion_tracking":      0,
        "cry_detection_switch": 0,
        "humanoid_filter":      1,
        "loudspeaker_vol_pct":  100,
        "jingle_mode":  0,
        "jingle_sound": 1,
        "jingle_volume":        100,
        "jingle_exist": 0,
        "flight_bright_mode":   0,
        "flight_light_brightness":      100,
        "flight_pir_set":       0,
        "flight_pir_one":       0,
        "flight_pir_two":       0,
        "flight_pir_three":     0,
        "flight_pir_sensitivity":       0,
        "flight_alarm_fun_onoff":       0,
        "flight_on_off":        0,
        "flight_pir_light_on_time":     30,
        "flight_warn_switch":   0,
        "flight_dualbrite":     0,
        "flight_ontime":        0,
        "flight_highbrightess": 10,
        "flight_lowbrightess":  10,
        "flight_mode":  0,
        "flight_motion_sens":   0,
        "onvif_enable": 1,

MonkeyMan007, 2022-08-06:
also it seems to reboot every 10 min

guino, 2022-08-06:
@MonkeyMan007 are you streaming the video when it reboots ? there's a known issue where streaming over UDP for 13 minutes causes the device to reboot -- if you stream over TCP this does not happen. 

I suggest you do this:
1-REMOVE the SD card and reboot -- this will make your device boot with original/OEM firmware -- if your device is rebooting every 10 minutes without the SD Card then you have a hardware problem (and should exchange/return it).
2-Try to work out the wiring for your CHIME without the SD card -- if you can't get it to work then either your device settings are incorrect OR your CHIME is incompatible OR your device is defective. The settings in the TUYA app (which is what you say you are using) should look like this (check BOTH Bell Settings and Chime Settings):
[image]
[image]
3-Verify you installed the chime device correctly -- this is done INSIDE your CHIME. You should have a little thing that looks like this:
[image]
Which gets installed like this:
[image]
If you don't have it or didn't connect it, then your CHIME will not work when pressing the bell button.