guino/ppsapp-rtsp #43: New hardware.... thanks Merkury

dmidthun, 2022-09-26:
Bought some new cams today, out of the box the base root "half" works: 

{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"<redacted>","softwareversion":"5.2.5","hardwareversion":"M11S_A5_V10_MIS1","firmwareversion":"ppstrong-a5-tuya2_general-5.2.5.20211201","identity":"","authkey":"DAAtLEvS6SxPQxN6VuvXTufTeLe0s66A","deviceid":"tstjb6f11613b631505d","pid":"aaa","WiFi MAC":"9c:1c:37:4c:a7:ae","ETH MAC":"9c:1c:37:4c:a7:ae"}

looking to get rtsp / onvif for these guys

### EDIT ###

Just noticed by chance that the models i picked up today are actually MI-CW217-101WWs, not MI-CW017-101Ws

Thanks Guino!!

guino, 2022-09-27:
@dmidthun is the deviceinfo the only thing that worked ? can you check if the /proc/cmdline works ? Some people have had success with 5.2.x versions using https://github.com/guino/Merkury1080P#conclusion -- have you tried that one yet ?

dmidthun, 2022-09-29:
The cmdline works as well, yes. I havent tried the 5.2.x versions as I may have overlooked that

guino, 2022-09-29:
@dmidthun the link I posted is the process that has worked for some people with 5.2.c firmware — yes, you use the 3 files posted with the steps from the previous project. That is the only process I heard has worked for some 5.2.x firmware.

dmidthun, 2022-09-29:
@guino I did try copying the 3 different files onto the card with no success. I am also in the process of attempting to figure out how to get a UART console to it without mucking up the pads too much with a crappy soldering iron

guino, 2022-10-04:
@dmidthun these boards are pretty fragile, even with a good soldering iron you can damage them if you're not careful. That said if you get the UART working let me know and I can give you some pointers. You can email me directly in that case (my email is on my github profile).