guino/ppsapp-rtsp #44: Patch Request - 2.7.12 Firmware for 720p

dmidthun, 2022-09-29:
I tried following the instructions to patch myself, but Ghidra kept crashing on both my linux and win machines... please let me know if you need anything else. Just RTSP is needed, as I dont think Onvif is available, and I dont use the cgi scripts.

**/devices/deviceinfo**:

{"devname":"Smart Home Camera","model":"Mini 7C","serialno":"061512432","softwareversion":"2.7.12","hardwareversion":"M7C_AK_V10_1245","firmwareversion":"ppstrong-a2-tuya2_geeni-2.7.12.20220621","authkey":"<redacted>","deviceid":"<redacted>","identity":"<redacted>","pid":"aaa","WiFi MAC":"dc:bd:7a:82:46:6b"}

**/proc/cmdline**:

mem=64M console=ttySAK0,115200n8 mtdparts=spi0.0:256k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,2496k(sys),4608k(app),640k(cfg) ppsAppParts=5 ip=0 - ip=30;/mnt/mmc01/initrun.sh)&:::::;date>/tmp/hack;(sleep

[ppsapp.txt](https://github.com/guino/ppsapp-rtsp/files/9679462/ppsapp.txt)

guino, 2022-10-04:
@dmidthun you may want to check the file -- the one you attached is 778Kb when most ppsapp files are 2MB+ in size. Maybe it didn't upload correctly because it was posted as s .txt (you should try just zip it and post it). If by any chance the file on the SD card is the same size (778Kb) then you may have removed the SD card too early after the first boot (before it could write the entire file) -- in which case you just have to delete the 'home' folder from the SD card and reboot to recreate a new home folder with a fresh copy of ppsapp in it. The file size is likely why ghidra was crashing.

dmidthun, 2022-10-05:
Not sure why. Did what you suggested and have the file... I tried following the instructions to patch on my own, but I think Tuya / Merkury has made a major revamp to their code and I am not savvy enough. (laughs). attaching the ppsapp that I got. changed cameras and SD cards 3 times and I got ghidra to work, so I know this one is good.
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/9714143/ppsapp.zip)

fw version 2.7.12

guino, 2022-10-05:
@dmidthun this one seems the right size I'll take a look and let you know.

guino, 2022-10-05:
@dmidthun you can try the patch below with these instructions: https://github.com/guino/ppsapp-rtsp/issues/1

[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/9717296/ppsapp-rtsp.zip)
snap/mjpeg address: 0x02c17a8
play.cgi address: 0x02c1fb4

ZikangXiong, 2023-02-15:
I followed #1 and had this file generated.
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/10739168/ppsapp.zip)

I can hear 2 beeps after booting, but can neither access by `rtsp://ip:8554` nor `http://user:password@ip:8080/cgi-bin/mjpeg.cgi` (where the address is updated and the httpd.conf keeps the user:password as default)

guino, 2023-02-16:
@ZikangXiong can you post a zip of your SD card contents (without the SDT folder) ? your ppsapp seems to be patched correctly. Most of the time the issue is something with wrong filename, location or wrong file.

ZikangXiong, 2023-02-17:
[genni-cam.zip](https://github.com/guino/ppsapp-rtsp/files/10762342/genni-cam.zip)

Here is the structure. 

```
.
├── bin
│   ├── cmd_router
│   ├── iwconfig
│   ├── lookup_proc
│   └── ppsconfig
├── busybox
├── cgi-bin
│   ├── cleanup.cgi
│   ├── main.cgi
│   ├── mjpeg.cgi
│   ├── snap.cgi
│   └── upload.cgi
├── custom.sh
├── env
├── etc
│   └── ssv6x5x-wifi.cfg
├── home
│   ├── ASC16
│   ├── anyka
│   │   ├── ak_info_dump.ko
│   │   ├── akcamera.ko
│   │   ├── isp_1034_dvp_101602.conf
│   │   ├── isp_h62.conf
│   │   ├── isp_sc1245.conf
│   │   ├── loadanyka3918
│   │   ├── sensor_gc1034.ko
│   │   ├── sensor_h62.ko
│   │   └── sensor_sc1245.ko
│   ├── app
│   │   ├── network
│   │   ├── ppsapp
│   │   └── ppsdsry
│   ├── ca.crt
│   ├── drv
│   │   ├── 8188fu.ko
│   │   ├── Strnio.ko
│   │   ├── atbm603x_wifi_usb.ko
│   │   ├── motor.ko
│   │   ├── otg-hs.ko
│   │   └── ssv6x5x.ko
│   ├── init.d
│   │   ├── S00config
│   │   ├── S01loadpps
│   │   ├── S20cmd_router
│   │   └── initS
│   ├── platform.env
│   └── sound
│       ├── dingdong.wav
│       ├── login.wav
│       └── restart.wav
├── hosts
├── httpd.conf
├── index.html
├── initrun.sh
├── jpeg-arm
├── lib
│   ├── libak_mt.so
│   ├── libakaudiocodec.so
│   ├── libakaudiofilter.so
│   ├── libakispsdk.so
│   ├── libakmedia.so
│   ├── libakstreamenc.so
│   └── libakuio.so
├── mqtt_pub
├── offline.sh
├── passwd
├── ppsFactoryTool.txt
├── ppsMmcTool.txt
├── ppsapp
├── ppsapp:Zone.Identifier
├── set
└── upload.html
```

guino, 2023-02-17:
@ZikangXiong First issue is your busybox was not downloaded correctly. Please download it from this link: 
https://github.com/guino/Merkury720/blob/main/mmc/busybox?raw=true

Next your snap/mjpeg.cgi files have `ADDRESS=0x02c17a8` please try just `ADDRESS=2c17a8` (not sure if this is an actual problem)

Your ppsapp file seems fine (patched correctly and in the correct location), so rtsp://ip:8554 should work , but please note this is not a **web browser** URL -- you have to use an application like VLC to play it. Also note it has been reported that VLC for macOS sometimes fails to play the rtsp feed from these devices. You can always download an app for RTSP on your phone and try it that way. Make sure there's no user/password for the RTSP url.

I would suggest fixing the busybox and testing the mjpeg first (http://user:password@ip:8080/cgi-bin/mjpeg.cgi as you described above), that's just so we know for sure the IP didn't change on you and that the files are being used.

I did take another look at the patch and looks correct as far as I can tell but I don't have a device on firmware 2.7.12 to test it.

Let me know how that goes.

ZikangXiong, 2023-02-17:
It works like a charm now. 

I suppose that the busybox is the issue, should have read your instruction more carefully. 

Thank you for such detailed explanation and remove the file containing my Wi-Fi password :-)