guino/ppsapp-rtsp #46: Patch request - Feit Electric Smart Indoor Camera, FW 4.0.8

torindkflt, 2022-10-08:
Contents of http://IP:8090/proc/cmdline (after already applying the hack):
`mem=64M console=ttySAK0,115200n8 loglevel=10 mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=0 - ip=30;/mnt/mmc01/initrun.sh)&:::::;date>/tmp/hack;(sleep`

Contents of http://IP:8090/devices/deviceinfo:
```
devname	"Smart Home Camera"
model	"Mini 8S"
serialno	"064257600"
softwareversion	"4.0.8"
hardwareversion	"M8S_A2_V10_F37"
firmwareversion	"ppstrong-a3-tuya2_feit-4.0.8.20210806"

identity	"MR2010090700913280"
authkey	"uo4CcxMbifDYN5KTXVkpCUAstbj1NSUD"
deviceid	"pp014b7b6e5067b7c4aa"
pid	"aaa"
WiFi MAC	"b4:fb:e3:16:5d:54"
ETH MAC	"00:00:00:00:00:00"
```

Original unpatched ppsapp is attached, MD5=`ab4e7e6afdaca800e0124e106154a2a6`
[ppsapp.zip](https://github.com/guino/ppsapp-rtsp/files/9738479/ppsapp.zip)

guino, 2022-10-08:
@torindkflt did you try editing the tuya_config.json file like mentioned here: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-745618597 ? or are you looking for snap/mjpeg, play.cgi addresses ?

guino, 2022-10-08:
@torindkflt here are the addresses:
snap/mjpeg.cgi: 0x038c300
play.cgi: 0x38d100

guino, 2022-10-08:
@torindkflt this firmware does not have the standalone RTSP code -- only onvif (whoch can probably be enabled by tuya_config.json) -- if for some reason that doesn't work let me know and I can try to force-enable it with a patch.

torindkflt, 2022-10-08:
I'm unable to telnet into it to modify tuya_config.json, it seems as if the separate busybox download won't run or is being blocked (I even tried moving the commands to start telnet from custom.sh to initrun.sh in case for some reason custom.sh wasn't running, and it still didn't work).

guino, 2022-10-08:
@torindkflt try this busybox: https://github.com/guino/Merkury720/blob/main/mmc/busybox?raw=true (size 1109128 bytes)

torindkflt, 2022-10-09:
Yes, that worked. Also, I didn't even need to modify tuya_config.json, instead I just uncommented the "/mnt/mmc01/set onvif_enable 1" command in custom.sh and that enabled it. Got the camera working with iSpy now. Thank you!