guino/ppsapp-rtsp #47: Is it possible to disable ssl/tls validation?

tosiara, 2022-12-19:
I'd like to inspect the traffic camera sends. Maybe you have tried that before? Is it possible?

```
[01-01 02:00:19-- TUYA Debug][mqtt_client.c:1139] select mqtt host:m2.tuyaeu.com
[01-01 02:00:19-- TUYA Notice][mqtt_client.c:1152] mqtt get serve ip success
[01-01 02:00:19-- TUYA Debug][mqtt_client.c:816] serverIP -1062731263 port:8883
[01-01 02:00:19-- TUYA Debug][mqtt_client.c:828] mqtt over TLS is enabled. Host:m2.tuyaeu.com Port:8883
 4704:0x8cb190: x509_verify_cert() returned -9984 (-0x2700)
[01-01 02:00:19-- TUYA Err][tuya_tls.c:941] mbedtls_ssl_handshake returned ffffd900 -9984
[01-01 02:00:19-- TUYA Debug][cloud_operation.c:422] replace CA for url m2.tuyaeu.com
[01-01 02:00:19-- TUYA Err][cloud_operation.c:434] can not find the replace CA
[01-01 02:00:19-- TUYA Err][mqtt_client.c:834] tuya_tls_connect error:-9984
[01-01 02:00:19-- TUYA Err][mqtt_client.c:1168] mqtt socket connect fail. retry_cnt:1 op_ret:-9984 errno:-100
[01-01 02:00:20-- TUYA Notice][mqtt_client.c:1320] MQ_EXIT_ERR : mqtt close
[01-01 02:00:20-- TUYA Debug][mqtt_client.c:695] close mqtt
[01-01 02:00:20-- TUYA Debug][mqtt_client.c:1339] mqtt fail_cnt:2 and sleeptime:5141 ms
```

```
[01-01 02:01:06:784 TUYA Err][http_inf.c:47] Open session failed: https://a2.tuyaeu.com/d.json?.... (-9984)
[01-01 02:01:06:815 TUYA Err][iot_httpc.c:926] Post Data Return Fail.-706
[01-01 02:01:06:824 TUYA Err][tuya_ipc_cloud_storage.c:175] httpc_ipc_cloud_storage_event_notify fail!
[01-01 02:01:06:842 TUYA Err][tuya_ipc_cloud_storage.c:587] fail to notify cloud start, ret -803
 4704:0x178f618: x509_verify_cert() returned -9984 (-0x2700)
[01-01 02:01:07:193 TUYA Err][tuya_tls.c:941] mbedtls_ssl_handshake returned ffffd900 -9984
[01-01 02:01:07:202 TUYA Err][cloud_operation.c:434] can not find the replace CA
```

guino, 2022-12-19:
@tosiara at a quick look at 2.7.x firmware we could probably make a patch to allow it to connect to the mqtt server without SSL. If you tell me the firmware you're using I can try to make a patch for you to test.

tosiara, 2022-12-19:
I'm using 5.2.1 and this is ppsapp: https://github.com/guino/Merkury1080P/files/10140862/squashfs-root.zip

I was thinking to make `x509_verify_cert` always return `0`, but unfortunately I'm not that much proficient with Ghidra.

Also, I tried to change all `m2.tuyaeu.com` and `a2.tuyaeu.com` to my controlled domain names using hex editor, but it still tries to connect to them, don't know where does it take those domains from

guino, 2022-12-19:
@tosiara to redirect the URL your best/simplest option is to edit /etc/hosts so it overrides the IP that the server resolves onto, like we do [here](https://github.com/guino/Merkury720/files/7672114/hosts.zip) but with the IP of your server (see relevant info here too: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-the-device-OFFLINE-(disconnected-from-tuya-servers)-%3F). 

It seems easier in the code to modify it to NOT use https/ssl at all than to try to force it to 'think' the certificate is good, so that's what I would recommend trying first.

tosiara, 2022-12-19:
I can redirect all the traffic without any problem, but ppsapp fails to validate my ssl cert. So, I'm looking for a way to either force use of plain HTTP or patch cert validation function to always bypass. Will read you provided link meanwhile

tosiara, 2022-12-20:
Scrolled briefly through that enormous "Off-cloud" thread. Wanted to try to patch that while loop that may be blocking other services when there is no internet ([comment](https://github.com/guino/BazzDoorbell/issues/4#issuecomment-741481814)). I couldn't find the main function in ghidra as per your instructions. The closest thing is this one, but it is grayed out and has no references:

[image]

Also, I patched ppsapp to contain http:// only urls, but ppsapp still uses the original urls. Where does it take them from???

```
$ strings ppsapp_patched | grep tuyaeu
http://a2.tuyaeu.com/d.json
https://a2-weaz.tuyaeu.com/d.json
*.tuyaeu.com1
*.tuyaeu.com1
*.tuyaeu.com1
*.tuyaeu.com1
```
```
[root@Meari /mnt/mmc01]$ ./ppsapp_patched
...
[01-01 02:00:11:473 TUYA Debug][mqtt_client.c:828] mqtt over TLS is enabled. Host:m2.tuyaeu.com Port:8883
...
[01-01 02:00:14:595 TUYA Debug][iot_httpc.c:920] Post URL: https://a2.tuyaeu.com/d.json?
```
Could you please help me to force plain HTTP for both MQTT and API? :pray:

tosiara, 2022-12-20:
UPDATE

I managed to solve this problem. Now I'm able to see all the traffic that the camera sends over SSL/TLS (MQTT and HTTPS). This can be done by patching `ppsapp` using hex editor.

1. Search for DER certficate header to get the offset (ex, 30 82 07 ce 30 82 06 b6 02 09 00 8c 7e 5f 76 e4). In my case it was at offset 0x2d8150 (decimal 2982224)
2. The certificate is 2002 bytes long
3. Generate your own cert to be used with ZAP or Burp or any other mitm proxy
4. Replace it

You should now see the API traffic:

[image]

And I can confirm that only MQTT is required for camera to startup and enable RTSP. So, I will try wirite a dummy MQTT server that allow camera to register itself with fake responses and continue booting.

guino, 2022-12-20:
@tosiara 
1. Sorry I didn't get to review and make a patch for your ppsapp before you figured it out, I've been a bit sick recently.
2. Nice work on getting the application to work with your server -- this is probably a better solution than patching code anyway.
3. If there's anything you're still waiting from me, please remind me so I don't forget it.

tosiara, 2022-12-21:
I replace the patched ppsapp in the squashfs and flashed it back. Also did the following change to `initrun.sh`:
```
#!/bin/sh

echo "192.168.2.1  m2.tuyaeu.com" > /etc/hosts
echo "127.0.0.1    a2.tuyaeu.com" >> /etc/hosts

sleep 30 && killall -9 ppsapp && sleep 5 && /opt/pps/app/appfiles/app/ppsapp &
...
.
```
192.168.2.1 is my raspberry in the same network that will run a fake MQTT server (in progress)

With this config the camera starts offline and RTSP is available

closed 2022-12-21 by tosiara

tosiara, 2022-12-23:
Made a successful PoC: https://github.com/tosiara/fake_mqtt_tuya
Now the camera boots into local RTSP completely offline :partying_face:

guino, 2022-12-24:
@tosiara this is a pretty cool project - I wonder if we can’t make a separate application to replace the CA certificate in the running ppsapp (memory) without having to kill (or even patch) it. I expect it should not be hard to automatically find and replace the CA certificate in memory so it works with any ppsapp version.

xraive, 2023-06-11:
@guino have you had a chance to work on this project?  It would be great to make these devices completely offline.

guino, 2023-06-14:
@xraive I have not worked and have no plans of working on this. I just threw the idea out there in case anyone wanted to pursue it.

samezrp, 2024-06-13:
@tosiara, I can see you have hardcoded certificates to application. I extracted mitm_srv.crt from ssl_server.c and when converted to DER it is 827 bytes long. Would you mind sharing DER certificate you were mentioning in:
[https://github.com/guino/ppsapp-rtsp/issues/47#issuecomment-1359460388](url)