guino/ppsapp-rtsp #50: RTSP Problem :(

MathisLec, 2023-04-27:
Hello there! I just root my cam which is 
```
"model": "Mini 9S",
"serialno": "106100343",
"tp": "100000017",
"hardware_version": "M9S_A2_V10_F37",
"software_version": "4.0.6",
"firmware_version": "ppstrong-a3-s_arenti-4.0.6.20210304"
```
  With onvif_enable activated.
  I have access to the telnet server where I can get netstat:
```
/ # netstat -ln
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 0.0.0.0:8000            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:8554            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN      
tcp        0      0 0.0.0.0:8090            0.0.0.0:*               LISTEN      
netstat: /proc/net/tcp6: No such file or directory
udp        0      0 0.0.0.0:3702            0.0.0.0:*                           
udp        0      0 0.0.0.0:3703            0.0.0.0:*                           
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
Active UNIX domain sockets (only servers)
Proto RefCnt Flags       Type       State         I-Node Path
unix  2      [ ACC ]     STREAM     LISTENING        635 /var/systemCmd.socket
```

which is great! (8554 port listening)

But when I try to get the rtsp stream in vlc
(```vlc rtsp://admin:admin@192.168.1.193:8554/Streaming/Channels/101```)
The ```--rtsp-tcp``` option don't exist for me
```vlc: unknown option or missing mandatory argument `--rtsp-tcp'```

I get an error, not very explicit, so I tried to look at the packages by wireshark:
[image]
Where 192.168.1.50 is my PC and 192.168.1.193 is my camera.

The error is not that clear.
Maybe I have the wrong URL, so I saw [this](https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826) post to get the URLs but when I start the tool, I get those errors:
```
./ipconvif 192.168.1.193                                                                            
[2023-04-27 23:08:16] [main.cpp] [26] {error:28 faultstring:Connection refused faultcode:SOAP-ENV:Sender faultsubcode:(null) faultdetail:connect failed in tcp_connect()}
[2023-04-27 23:08:16] [main.cpp] [26] {error:28 faultstring:Connection refused faultcode:SOAP-ENV:Sender faultsubcode:(null) faultdetail:connect failed in tcp_connect()}
[2023-04-27 23:08:16] [main.cpp] [26] {error:28 faultstring:Connection refused faultcode:SOAP-ENV:Sender faultsubcode:(null) faultdetail:connect failed in tcp_connect()}
[2023-04-27 23:08:16] [main.cpp] [26] {error:28 faultstring:Connection refused faultcode:SOAP-ENV:Sender faultsubcode:(null) faultdetail:connect failed in tcp_connect()}
```
So I don't really know what to do, when I copy paste this error on google, it is not really relevant for me...
Any idea on your side?

**EDIT:** I just tried on [RTSP Player](https://gardinal.net/rtsp-player/) and it worked! (but very slowly and in black and white, any optimization to do?) I don't really know why it is not working with vlc... and ipconvif don't works yet.

guino, 2023-04-28:
@MathisLec what version of VLC are you running ? it works for me:
```
$ vlc --rtsp-tcp rtsp://10.10.10.88:8554
VLC media player 3.0.18 Vetinari (revision 3.0.13-8-g41878ff4f2)
[000055625166d5d0] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00007f7790002260] gl gl: Initialized libplacebo v5.229.2 (API v229)
```

The ONVIF implementation on these devices is known to be 'flawed' and/or incompatible with many devices/applications (VLC for MacOS has been reported as having issues in the past).  
The URL rtsp://admin@admin@IP:8554/Streaming/Channels/101 should work but you can also try just rtsp://IP:8554/Streaming/Channels/101 and VLC should ask for user/password when it starts (if needed).

You can try a phone app for RTSP/ONVIF or even just a different VLC version and/or different machine. If it worked with RTSP player the device is working (however it can). The 102 channel is lower resolution but should work as well (faster for preview). Both channels should be in color unless the night-vision is enabled on the device (i.e. dark room).

guino, 2023-04-28:
I just tried VLC for mac 2.2.5.1 and it worked fine (and has the --rtsp-tcp option too) -- the only thing is: I don't have any device on 4.0.x firmware (I have only 2.7.x, 2.9.x, 2.10.x and 7.6.32), so the RTSP implementation may be different.