guino/ppsapp-rtsp #51: help needed pacth file, cant find ppsapp

irockX123, 2023-07-16:
{"devname":"Smart Home Camera","model":"Bullet 4S","serialno":"064868801","softwareversion":"5.0.5","hardwareversion":"B4S_V10_S1_GC1","firmwareversion":"ppstrong-c71-tuya2_lsc-5.0.5.20210301","identity":"M1M001AA3202015266","authkey":"g4MVARVFmfmn0fWHp0ea2SIw20SqvUOq","deviceid":"pp015ad0114f79c3ea5e","pid":"aaa","WiFi MAC":"b4:fb:e3:fc:28:2c","ETH MAC":"b4:fb:e3:fc:28:2c"}

[image]

i cant find the ppsapp file can somebody help me?

guino, 2023-07-16:
@irockX123 I'm pretty sure 5.0.5 is one of the versions that is not running linux. If you post the output to the /proc/cmdline I can likely confirm that for you. We do not have any solution for the devices running RTOS. 

I haven't heard if 5.0.5 supports WebRTC (https://www.reddit.com/r/smartlife/comments/oyqvdv/webrtc_stream_terminal_for_tuya_smartlife_cameras/) -- If it does you can use it to get RTSP using something like this: https://github.com/koush/scrypted

irockX123, 2023-07-16:
hi Guino
here is what you ask for

console=/dev/null LX_MEM=0x3fe0000 mma_heap=mma_heap_name0,miu=0,sz=0x1d00000 pcbversion=B3S_S1_V10 sensor=gc2063mipi

guino, 2023-07-16:
@irockX123 unfortunately that is the non-linux firmware (RTOS), the tools/processes we published so far will not work on that firmware. I am unaware of any tools/mods for that firmware/OS at this time -- my best guess is any changes would require opening the device, getting a copy of the flash modifying it and flashing it back. Even then you'd only be able to use whatever code/function was included in it (i.e. RTSP if it the code is there but just 'disabled'), but the only way to find out is looking at the firmware code.