guino/ppsapp-rtsp #53: NO RTSP after patching.  Tried all three patches for  ppstrong-a2-tuya2_geeni-2.7.6.20210207 listed.  snap.cgi and mjpeg.cgi working correctly with the addresses provided. Patched files are the correct filesize, placed in root of SD and I am getting two tones when I reboot the camera. VLC stating connection refused used both UDP and TCP 554 and 8554 -Connection refused by peer. Port scanning corroborates this - 23,80,8080,3702 and 3703 open.

guino, 2023-08-28:
NO RTSP after patching.  Tried all three patches for  ppstrong-a2-tuya2_geeni-2.7.6.20210207 listed.  snap.cgi and mjpeg.cgi working correctly with the addresses provided. Patched files are the correct filesize, placed in root of SD and I am getting two tones when I reboot the camera. VLC stating connection refused used both UDP and TCP 554 and 8554 -Connection refused by peer. Port scanning corroborates this - 23,80,8080,3702 and 3703 open.  

{"devname":"Smart Home Camera","model":"Mini 7C","softwareversion":"2.7.6","hardwareversion":"M7C_AK_V10_1245","firmwareversion":"ppstrong-a2-tuya2_geeni-2.7.6.20210207","deviceid":"pp01a7848e810bf2315c","identity":"MR2006120101400530","pid":"aaa"}

MD5: d39b6dc4f80e3d6d8b131b5034f9e4ae

patches tried 

[ppsapp-offline.zip](https://github.com/guino/ppsapp-rtsp/files/12388354/ppsapp-offline.zip)
[ppsapp-rtsp.zip](https://github.com/guino/ppsapp-rtsp/files/12388355/ppsapp-rtsp.zip)
[ppsapp-rtsp-sd.zip](https://github.com/guino/ppsapp-rtsp/files/12388356/ppsapp-rtsp-sd.zip)

used https://www.marcrobledo.com/RomPatcher.js/ as instructed

Ive even tried extracting the patch archives and patching the .ips files directly - still no luck.

_Originally posted by @NigelSamhain in https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-1685168220_

guino, 2023-08-28:
@NigelSamhain can you post a zip of your SD card without the SDT folder for review ? You should only need the ppsapp-rtsp.zip patch to make it work. Sometimes the issue is something simple like wrong filename or file in the wrong location.

Placid-Storm, 2023-09-07:
Ever figure this out? I'm having the same issue with the same camera but a newer FW and dif checksum.

guino, 2023-11-13:
@Placid-Storm if you post a zip of your SD card files without the SDT folder I can review it when I have a chance -- I never received any more details on this issue so I could not follow up.