guino/ppsapp-rtsp #54: Not found in list

xNickXx, 2024-03-16:
hello, i have the MD5 but there is no match with those you have.
do you have the latest please  ?

81a9b446fbe2e01fa9c5124f6972f2bd
"ppstrong-c51-tuya2_tic-2.9.6.20200515"
"M8S_H1_V10_F23"
"Mini 8S"

guino, 2024-03-29:
@xNickXx sorry for the delayed response, can you post a zip of ppsapp file with the details you provided above ? I don't have a device with that specific ppsapp file so I need it in order to try and make a patch for it.

guino, 2024-04-15:
There was only 1 user I know that had 5.2.4 and was able to get it modded: https://github.com/guino/Merkury1080P/issues/23 -- the camera may be different model but you could check the information and try it out.