guino/ppsapp-rtsp #56: Hi,  Can you please patch my psapp for me?

RedScorpio83, 2024-09-11:
I've tried myself but I can't get it to work, can you please help me out?

Thank you

{
    "devname": "Smart Home Camera",
    "model": "Bell 5T",
    "serialno": "xxx",
    "softwareversion": "4.3.0",
    "hardwareversion": "BE5T_H1_V10_433",
    "firmwareversion": "ppstrong-c5-tuya2_joys-4.3.0.20200827",
    "identity": "MR2008150100100311",
    "authkey": "xxx",
    "deviceid": "pp014a3975ff37adfba7",
    "pid": "aaa",
    "WiFi MAC": "7c:25:da:6a:60:d9"
}

guino, 2024-10-01:
You need to post a zip of the ppsapp from your device on the github issue so I can take a look at it. That said,a lot of 4.x firmware will work without patching, see: https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826

Stormtracker221, 2024-12-23:
Holiday patch request- Globe (Bell) doorbell cam 2.10.6
info:  http://admin:056565099@192.168.1.146/devices/deviceinfo
{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"059836550","softwareversion":"2.10.6","hardwareversion":"BE8S_H1_V10_915","firmwareversion":"ppstrong-c51-tuya2_globe-2.10.6.20210813","authkey":"DMc7FYfZK4h7KAFpxLlkXuZzA8utlsmN","deviceid":"pp0177bd401c01f86543","identity":"MR2004060101401546","pid":"aaa","WiFi MAC":"d4:d2:d6:12:50:56"}

Thank you and Merry Christmas