guino/ppsapp-rtsp #57: Request patch-Globe (Bell) doorbell cam 2.10.6

Stormtracker221, 2024-12-23:
Holiday patch request- Globe (Bell) doorbell cam 2.10.6
info: http://admin:056565099@192.168.1.146/devices/deviceinfo
{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"059836550","softwareversion":"2.10.6","hardwareversion":"BE8S_H1_V10_915","firmwareversion":"ppstrong-c51-tuya2_globe-2.10.6.20210813","authkey":"DMc7FYfZK4h7KAFpxLlkXuZzA8utlsmN","deviceid":"pp0177bd401c01f86543","identity":"MR2004060101401546","pid":"aaa","WiFi MAC":"d4:d2:d6:12:50:56"}

Thank you and Merry Christmas

guino, 2024-12-23:
@Stormtracker221 You need to post a zip of the ppsapp from your device on the github issue so I can take a look at it.

Stormtracker221, 2024-12-24:
Will do.Thank you for your prompt response.

Stormtracker221, 2024-12-24:
Zip as requested as requested. 
[ppsapp.zip](https://github.com/user-attachments/files/18235145/ppsapp.zip)
Let me know if you need any other info.

Thank you

Stormtracker221, 2024-12-24:
FYIIt took me a while but I did get to the point where it said "done", (hack completed)However, after that I went to try and set up another blank sd card using the same process, including holding the reset button on start up. That card didn't work, but my concern was that my holding the reset button again may have screwed up the flash process when I did the first sd card.
Let me know if you need any other info.
Thanks again

guino, 2024-12-24:
@Stormtracker221 I have looked at the provided ppsapp and it should support onvif with url rtsp://ip:8554/Streaming/Channels/101 like many other devices running 2.10.x without any patching. For this simply change the line #/mnt/mmc01/set onvif_enable 1 to /mnt/mmc01/set onvif_enable 1 (just remove the # ) in custom.sh -- **and** copy the ppsapp file to the root of the SD card (without patching it).

If this doesn't work for some reason let me know.

Stormtracker221, 2024-12-25:
So I did already remove the # form that line to enable onvif, but still no go.I did notice now, after copying and pasting the ppsapp file to the root of the sd card, that I do get the 2 chimes on boot up, as it should (like my hacked and working Merkury 720 cam); It didn't that previously. A good sign ? But still cant connect using the rtsp line on vlcWhen I try using Ispy or run a onvif scanner it says "No media endpoints"  . If that helps.
I'm still trying.....Not giving upLet me know of I can provide more diagnostic info for you.
Thank you Merry Christmas or Boxing day